esp8266/Arduino #2435: WiFiTelnetToSerial Example - Minor Issues

nameoftherose, 2016-08-25:
Reference is made to [WiFiTelnetToSerial.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino)
The loop in lines 55-63 continues even if an available client slot has been found. As a result all empty spots are allocated and spurious _New Client_ messages are produced. I propose that the `continue` statement in line 61 is replaced by a `break`.
The statements in lines 65-66 are executed always even if a client spot was previously allocated. As a result, a race condition may arise and an incoming connection rejected. I propose that these statements are executed only when `i==MAX_SRV_CLIENTS` ie said lines 65-66 are replaced with:
````
if (i==MAX_SRV_CLIENTS) {
      WiFiClient serverClient = server.available();
      serverClient.stop();
}
````

devyte, 2017-10-14:
@nameoftherose The code you reference is still there in latest git.
Your changes seem sane to me. Did you test these modifications? If so, could you please make a PR?

nameoftherose, 2018-01-29:
@devyte I am sorry for the delayed response. Indeed I have tested these modifications. I did the PR as requested. Thank You.

closed 2018-01-29 by devyte