esp8266/Arduino #2438: OTA starts but fails, esp unreachable

supersjimmie, 2016-08-25:
Sketch uses 314,161 bytes (30%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 39,992 bytes (48%) of dynamic memory, leaving 41,928 bytes for local variables. Maximum is 81,920 bytes.
Uploading...........................................................................................................................................................................................................................
18:50:59 [ERROR]: No Result!

### Basic Infos
I have a laptop with Windows 10 in the same IP network segment and connected to the same wifi AP.
I do see the esp under Tools - Port.

#### Hardware
Hardware:			NodeMCU v2
Core Version:      	2.3.0

### Description

Upload starts but after a couple of seconds it stops.
Then [ERROR]: No Result!

Before, and partially during the upload I can ping the esp, after a few seconds it's no more pingable and it never comes back.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB
CPU Frequency:		80Mhz
Upload Using:		OTA / SERIAL
Reset Method:		nodemcu

### Sketch 
Stripped to the relevant parts, offcourse I connect to wifi and work with sensors and stuff.

```cpp

#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

void setup() {
  setupOTA();
}

void loop() {
  ArduinoOTA.handle();
}

void setupOTA() {
  ArduinoOTA.setHostname("SlimmeMeter");
  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println(F("\nEnd"));
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println(F("OTA Ready"));
}
```

### Debug Messages

Uploading...........................................................................................................................................................................................................................
18:50:59 [ERROR]: No Result!

DanielHaubjergVestergaard, 2016-08-26:
I have the same problem with upload OTA to my WeMos D1
---------------------------------------------------------------------------------------------------------
Authenticating...OK
Uploading.......................................................................
................................................................................
......................
19:25:36 [ERROR]: No Result!

supersjimmie, 2016-08-27:
@DanielHaubjergVestergaard Solved here, but I don't know why/how.
I enabled debugging to see what goes wrong and now it just works!
Can you try that too, just to see?

supersjimmie, 2016-08-27:
Oh, bummer, it looks like it works but it doesn't because it boots with the old program instead of the newly uploaded.

supersjimmie, 2016-08-27:
So now I think I know what's the difference between when it worked and when not.
If I keep usb connected (for logging) the esp boots fine, without usb connected it fails somewhere (but I cannot see where/why).

My module is powered externally with 5V on it's Vin pin and it normally works fine with that.
Even after a reset, or even after reconnecting usb and then a reset, it still won't work.
So it seems that someting fails during the ota process if no usb connected?

me-no-dev, 2016-08-27:
what do you get in Serial when it fails? Or do you say that it will not fail if Serial is attached?
Also are you using 2.3.0 or git version?

me-no-dev, 2016-08-27:
I just merged [this](https://github.com/esp8266/Arduino/pull/2445), which may help with your issues.

supersjimmie, 2016-08-28:
@me-no-dev as said, with serial connected it succeeds so I cannot see what happens when it fails.
That makes it very hard to figure out what happens when it fails. :(

I will download your modified ArduinoOTA.cpp/h files to try.

supersjimmie, 2016-08-28:
I got https://github.com/esp8266/Arduino/commit/889775c6fe3c23a390cfbb0b57e6de371292a361 and uploaded my sketch again (over serial) to make it running with those latest changes.
Then I tried an OTA again.
Slight difference it that I don't get the [ERROR] No result anymore, but it's still unreachable after OTA.
Then I only connected serial (which also causes a reset) and it boots fine.

Uploaded again over OTA, again dead.
Then I enabled debugging for OTA+Updater, uploaded over Serial, connected Putty to Serial, did an OTA update, and all goes fine.

> sleep disable
             [begin] roundedSize:       0x0004F000 (323584)
                                                           [begin] updateEndAddress:  0x00100000 (1048576)
                                                                                                          [begin] currentSketchSize: 0x0004F000 (323584)
                                                                                                                                                        [begin] _startAddress:     0x000B1000 (724992)
                                                                                                                                                                                                      [begin] _currentAddress:   0x000B1000 (724992)
                                 [begin] _size:             0x0004E510 (320784)
                                                                               Start updating sketch
MD5 Success: f919aef93f95b7e75e77a26025b55505
                                             Staged: address:0x000B1000, size:0x0004E510
                                                                                        Update Success

                                                                                                      End
Rebooting...
            state: 5 -> 0 (0)
                             rm 0
                                 pm close 7
                                           del if0
                                                  usl

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
@cp:0
     ld
       Start.

supersjimmie, 2016-08-28:
btw, during the last SUCCESS OTA, the IDE says:
Uploading............................................................................................................................................................................................................................An error occurred while uploading the sketch

me-no-dev, 2016-08-28:
is it possible that you have tested the following way:
- Uploaded the sketch over Serial
- Tried OTA without resetting the ESP manually first (because after Serial upload, if not manually reset, ESP.restart() will actually hand the ESP)
- You uploaded with debug enabled
- You connected to the ESP after the upload with putty which as you said reset the ESP
- Tested OTA and everything went well

If this is what you did, the fix and explanation is in the second point :)

supersjimmie, 2016-08-29:
No it is not a fix, to make it short: OTA only works with Serial connected.
Which is, offcourse, pretty useless, because I want OTA without Serial connected...
In any way and any order that I tried, OTA does not work without Serial connected.

supersjimmie, 2016-08-29:
Yes I believe you that it works with you.
But as with more issues, there may be some cause why it doesn't always work everywhere else.  ;)

me-no-dev, 2016-08-29:
Did you try the exact steps above? It's really, really important to manually reset after Serial upload for OTA to work!

supersjimmie, 2016-08-29:
Ah, I understand., thanks for clarifying.
No I did not.  But my esp always automatically reboots after a Serial upload, that's not enough then?  I think I can recall some kind of issue with that restart() now...
(will try later in the evening)

igrr, 2016-08-29:
ESP doesn't automatically reboot after Serial upload. It starts running program directly. So if you are using OTA function you need to do a manual restart (press reset button or toggle power).
Depending on how you have DTR and RTS wired from USB-Serial converter to the ESP, it may be resetting each time you open serial terminal window.

supersjimmie, 2016-08-29:
Good to know @igrr , that seems to explain.
(The NodeMCU has onboard CH340 so it reboots even when I connect that USB to my laptop)
I thought it was a reset after Serial upload, but it isn't.
Will try asap and if that's why it didn't work I can close this. :+1:

supersjimmie, 2016-08-29:
@igrr @me-no-dev thanks, solved now I do a reset first thing after a Serial upload.
(I still have some issue that my esp does not always boot without serial but that's not OTA related and probably has to do with the required low/high gpio's)

closed 2016-08-29 by supersjimmie

glyndon, 2017-01-16:
The Wiki  page for the ArduinoOTA library doesn't make mention of the need for a hard reset after serial upload of an OTA-containing sketch. 
None of the third-party blogs on the subject do either.
Seems like an important step (it fixed the problem for me), worth adding to the docs.

2000jago, 2017-12-07:
I am having this issue also but the suggestions above have not solved it for me.  Having to do a manual reset after an OTA upload makes OTA uploading pretty pointless in my situation as I don't have (easy) access to the device as it's deployed within waterproof housing outdoors, some distance from my computer, hence the need for OTA uploads.  
Any other suggestions?

devyte, 2017-12-07:
@2000jago If you read carefully, the suggestion above for a physical reset is meant only for right after update over serial. It is not meant for right after update OTA.
I suggest using the OTA events, and on start close Serial, then once done open Serial again, if you even need Serial.

2000jago, 2017-12-08:
Noted.  I do not need serial, and have indeed hard reset the device after updating over USB.  (Naturally since it has to be unplugged from USB and relocated to it's deployed location where it is powered by battery/solar once connected.)  But the problem is that after an OTA update (which is indeed successful), the wemos does not reset itself and just hangs.  It requires power cycle to resume operation with the new code sent via OTA.  At which point it is fine until a new update is sent via OTA.  I then have to power cycle the device to resume which is very inconvenient.

davepl, 2019-12-06:
Two random comments:
- When WiFi stuff fails without USB connected I often suspect 3.3v rail power
- Make sure you're calling SPIFFS.end() in the onBegin handler of SetupOTA