esp8266/Arduino #2439: Get time

panagorn, 2016-08-25:
Is it possible to get the current time without using any hardware or internet connection?

supersjimmie, 2016-08-25:
Enter current time after startup over Serial?

panagorn, 2016-08-25:
my esp is connected with battery

supersjimmie, 2016-08-26:
So your esp has really nothing: no internet, no serial, no other input not any way to reach anything?
Then it is impossible.  It has no builtin clock so when it is booted it has no idea about time.

panagorn, 2016-08-26:
If it has internet is it possible to get it one time at boot and then know the time without any other request?

everslick, 2016-08-26:
yes, but the internal software clock will drift over time. here i see a drift of around 160ms per hour.

supersjimmie, 2016-08-26:
Since you say it runs on a battery, I think it is not running for too many hours without a reboot (deep sleep).
So if you do have internet and read a time source over internet at boot, you will be able to use that time for hours if you don't mind losing perhaps a second.

panagorn, 2016-08-27:
@supersjimmie how can i do that?

supersjimmie, 2016-08-28:
Just do a qck search for the Time library?

perryc, 2016-10-01:
It was not the primary reason to include it in the design but I set time from a cheap GPS chip.

devyte, 2017-10-14:
@panagorn There are RTC modules that keep time. They all have some drift, although minimal.
Alternatively, there is TimeLib, which can keep time using the internal cpu counter after you set it. It will drift slightly over time, you'll want to sync via NTP once in a while (e.g.: once a day, once every few days), You can use e.g.: NtpClientLib to sync via NTP. 
Closing per #3655 .

closed 2017-10-14 by devyte