esp8266/Arduino #2441: ESP in dual mode, AP and station

ranjit84, 2016-08-26:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

WereCatf, 2016-08-26:
About HTML-pages not being served whole: I've seen several people complaining about similar issues lately and I think it's a bug with the ESP8266-core itself, not in your code. Have you tried an older version of the core, like e.g. 2.2.0, to see if it works any better?

devyte, 2017-10-14:
@ranjit84 The ESP has a single radio, shared between STA and AP. In dual mode, the SoftAP must be set to the same channel as the STA that you connect to. If you don't, once the STA connects, the SoftAP will be dragged along, and you'll lose connectivity.
In addition, there are some issues with serving webpages with ESP8266WebServer, especially when the webpages have embedded elements. In such cases, modern browsers tend to get creative and try to retrieve concurrenly.
Assuming the issue is still relevant, please try:
- latest git
- PR #3215 (sdk 2.1.0)
- PR #3362 (lwip2)
- ESPAsyncWebServer (an almost drop-in replacement webserver, which is what I currently use without issues)
Please report back here what works.

devyte, 2018-01-08:
Closing due to lack of feedback in over 2 months, and age.

closed 2018-01-08 by devyte