esp8266/Arduino #2446: Running ESP8266 Serial WiFi example on Arduino Uno

titusece, 2016-08-27:
Hi,

I've ESP8266 Serial WiFi module, want to run some wifi examples using Arduino UNO board, I did the following but not able to find the ESP8266 based WiFI examples after restart the arduino IDE.

Can you please help me out of this problem ?

cd hardware
mkdir esp8266com
cd esp8266com
git clone https://github.com/esp8266/Arduino.git esp8266
cd esp8266/tools
python get.py

WereCatf, 2016-08-27:
All this code is meant for running the ESP8266 **itself** as an Arduino-device, not as a device to provide another Arduino with WiFi. If you want to use another Arduino and use an ESP8266 to provide WiFi for it you should flash the AT-firmware and look elsewhere for examples on how to use it with your Arduino.

titusece, 2016-08-29:
Thanks WereCatf

LuizHAP, 2016-08-31:
@WereCatf do u have another library for that?

WereCatf, 2016-08-31:
@LuizHAP No, I do not. Google for "esp8266 arduino uno" or something for examples on how to use it.

supersjimmie, 2016-09-03:
Take a look at https://github.com/sparkfun/SparkFun_ESP8266_AT_Arduino_Library
(don't let the picture mislead you, just look at the code, I think that is what you need)

closed 2016-10-27 by titusece