esp8266/Arduino #2447: Boot Issues and WiFi Client Connecting without WiFiClient.Begin Called

sk1tt1sh, 2016-08-27:
### Basic Infos

#### Hardware
Hardware:			ESP-12e
Core Version:      	?

### Description

I have discovered a issues with the ESP-
1. Program the module with a WiFi sketch that perhaps just connects to wifi - Watch network from router or w/e to see that ESP_**** device shows up and get the IP.
2. Program the module with a plain blink sketch only

What you will see is once you program the blink sketch without even including wifi libraries the module will reconnect to wifi without the wificlient ever being called.

Is this normal?

Second:
When programming the module with a plain blink sketch it will successfully boot when power is connected _every time_ however, if I add ANY libraries or functionality other than blink sketch it will not boot.  (This is where I discovered the above) - To add to the oddities it will boot if UART is connected, but will not boot if it is not. **Even without Serial ever being called and no DEBUG defines.** I find it strange that it doesn't boot without UART connected but as soon as I do it will. 

Is this normal?

Please help

### Settings in IDE

Module:  			Adafruit Huzzah
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		PlatformIO/Arduino defaults
Flash Frequency:	PlatformIO/Arduino defaults
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch
```cpp

#include <Arduino.h>

const char WiFiSSID[]        = "SSID";
const char WiFiPSK[]         = "PSK";

void setup() {
  connectWifi();
}

void connectWifi(){
  WiFi.mode(WIFI_STA);
  WiFi.begin(WiFiSSID, WiFiPSK);

  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
  }
}

void loop() {

}
```
```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>

void setup() {
  pinMode(2, OUTPUT);
}

void loop() {
  digitalWrite(2, HIGH);
  delay(1000);
  digitalWrite(2, LOW);
  delay(2000);
}
```

### Debug Messages

```
N/A
```

krzychb, 2016-08-28:
Hi @sk1tt1sh,

> What you will see is once you program the blink sketch without even including wifi libraries the module will reconnect to wifi without the wificlient ever being called.

This is default functionality. ESP reconness to the last used access point basing on credentials saved in flash. As you observed, this happens for a bare blink sketch without calling any Wi-Fi function or including libraries. To change this functionality you can use `WiFi.disconnect()`, `WiFi.mode(WIFI_OFF)` or `WiFi.setAutoConnect()`, see  https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/readme.md#station

> Second:
When programming the module with a plain blink sketch it will successfully boot when power is connected every time however, if I add ANY libraries or functionality other than blink sketch it will not boot. 

What do you mean it will successfully boot or not boot? How to check it?

I have loaded your second sketch with and without:
```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
```
and do not observe any change. LED is blinking for either version of this sketch.

sk1tt1sh, 2016-08-28:
Thanks for clearing that up with the autoconnect behavior. 

With booting and not booting:
I will get the IP of the esp and ping it constantly. Almost always when I plug it in I never see it connected and responding to pings. When I press the rst it will start up and connect to WiFi almost always. I have tried with and without the blinking led. At one point I thought perhaps it was because I was monkeying with gpio2 to flash the built in led but the test blink sketch uses it and successful power up every time I plug it in without the need to press the rst. When the device is doing this it will flash the leds on GPIO0 & GPIO2 every ~10 seconds like its trying to start. Sometimes it will even start too. As mentioned in the issue submission if I connect to uart to watch the boot messages it will start working.

sk1tt1sh, 2016-08-29:
I have discovered the cause to the issue.

I was using SoftwareSerial and it makes the `void ICACHE_RAM_ATTR sws_isr_2() { ObjList[2]->rxRead(); };` call prior to me setting GPIO2 as an output after boot and writing to it.

I apologize for neglecting to add that as part of the issue. Please see https://github.com/plerup/espsoftwareserial/issues/28 for more detail.

Thanks!

closed 2016-08-29 by sk1tt1sh

krzychb, 2016-08-29:
@sk1tt1sh,
I am happy you got this one to resolution and thank you for update!