esp8266/Arduino #2450: SPIFFS read speed varies depending on location in file

2dom, 2016-08-28:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.2

### Description

I experience good read speed over most of the file system. 

E.g. if I have a 2MB image file, the read speed is extremely slow in some parts of the file but ok in others.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/3MB SPIFFS
CPU Frequency:		160MHz
Flash Mode:			?
Flash Frequency:	?
Upload Using:		SERIAL
Reset Method:		Nodemcu

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

WereCatf, 2016-08-29:
Does the drop in performance happen around 2147483648 bytes read?

2dom, 2016-08-29:
The first drop happens at 2173912 and lasts a few 10 kBs ....

WereCatf, 2016-08-29:
You could try a newer SPIFFS from https://github.com/pellepl/spiffs -- it's at v3.6, whereas the SPIFFS in the ESP8266 Arduino-core is v3.4. If you do test it I'd like to hear your results. Also, if it doesn't resolve the issue you could open an issue in @pellepl 's code as he's much more likely to be able to tell you what the problem is than anyone here.

WereCatf, 2016-08-29:
The SPIFFS v3.6 has this note:
> Optimize SPIFFS_read for large files

WereCatf, 2016-08-29:
I just tested this and I can confirm that on SPIFFS v3.4 the performance-drop happens, but when I updated to v3.6 it disappeared -- no drops, whatsoever. My version of the ESP8266-core with SPIFFS 3.6 is available at https://github.com/WereCatf/Arduino if you wish to test.

2dom, 2016-08-29:
I tested it with the v3.6 version ... it works without slow down. Great stuff. Do you have any idea when the ESP8266-core will be updated. I stumbled over this issue twice now.

WereCatf, 2016-08-29:
Considering that, as far as I know, @igrr and whoever else are working on this core for free and on their own time I don't think they have any sort of a schedule or anything, so there's no way of saying when they might take a look at this. Just be patient, I'm sure someone will at least comment on this topic and/or the pull-request eventually.

2dom, 2016-08-30:
I am patient ... just asking. What you guys are doing is amazing - cheers to you!

devyte, 2017-10-14:
@2dom Current SPIFFS version is 0.3.7, so this is already done.
Closing as resolved.

closed 2017-10-14 by devyte