esp8266/Arduino #2453: Certificate doesn't match for letsencrypt cert.

zagaria, 2016-08-29:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

Certificate doesn't match for letsencrypt cert.

#### Hardware
Hardware:			ESP-12
Core Version:      	2.3.0

### Description

Certificate doesn't match for letsencrypt cert. Basic tested on host letsencrypt.org I tried on other hosts with LetsEncrypt cert, but get all the same message.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		 SERIAL
Reset Method:	 nodemcu

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <ESP8266mDNS.h>

const char* ssid = "....";
const char* password = "....";

const short int httpsPort = 443;
const char* host = "letsencrypt.org";
const char* fingerprint = "‎94 fc 72 4e 10 ba 82 72 fc 81 4b 65 0c e5 a6 48 52 bd e5 6e";

void GetContent();

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  GetContent();

}

void GetContent()
{
   WiFiClientSecure client;
  Serial.print("connecting to ");
  Serial.println(host);
  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

  if (client.verify(fingerprint, host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages

```
Connected to WiFi
IP address: 192.168.0.2
connecting to letsencrypt.org
certificate doesn't match

```

igrr, 2016-08-29:
Please enable debug output in the IDE (Tools > Debug Level > Core + SSL), add `Serial.setDebugOutput(true);` line after `Serial.begin` line, run the program again, and post full serial output here. Thanks.

zagaria, 2016-08-29:
I find error. It's my mistake. If I copy Thumbprint from browser (Chrome), it include some incorrect characters that not visible. Fix using "Fix encoding & Reload".

Sorry for false alarm.

closed 2016-08-29 by zagaria