esp8266/Arduino #2454: Is it possible to change ADC speed to speed up analogRead

ataytanzer, 2016-08-29:
### Basic Infos

#### Hardware
Hardware:		Wemos D1 mini (ESP8266)
Core Version:      	2.1.0

### Description

Problem description

### Settings in IDE

Module:  			Wemos D1 mini
Flash Size: 		4MB
CPU Frequency:		160Mhz
Flash Mode:			
Flash Frequency:	
Upload Using:		 SERIAL
Reset Method:		

### Sketch

#ifndef cbi
#define cbi(sfr, bit) (_SFR_BYTE(sfr) &= ~_BV(bit))
#endif
#ifndef sbi
#define sbi(sfr, bit) (_SFR_BYTE(sfr) |= _BV(bit))
#endif

// set prescale to 16
sbi(ADCSRA,ADPS2) ;
cbi(ADCSRA,ADPS1) ;
cbi(ADCSRA,ADPS0) ;
```

### Debug Messages
 sbi(ADCSRA,ADPS2) ;

     ^

C:\Users\tanze\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/pgmspace.h:31:23: note: in definition of macro '_SFR_BYTE'

 #define _SFR_BYTE(n) (n)

                       ^

C:\Users\tanze\Desktop\Spo2_650usec_wemos\Spo2_650usec_wemos.ino:43:1: note: in expansion of macro 'sbi'

 sbi(ADCSRA,ADPS2) ;

 ^

In file included from C:\Users\tanze\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Servo\src/Servo.h:52:0,

                 from C:\Users\tanze\Desktop\Spo2_650usec_wemos\Spo2_650usec_wemos.ino:2:

Spo2_650usec_wemos:43: error: 'ADPS2' was not declared in this scope

 sbi(ADCSRA,ADPS2) ;

            ^

C:\Users\tanze\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:184:25: note: in definition of macro '_BV'

 #define _BV(b) (1UL << (b))

                         ^

C:\Users\tanze\Desktop\Spo2_650usec_wemos\Spo2_650usec_wemos.ino:43:1: note: in expansion of macro 'sbi'

 sbi(ADCSRA,ADPS2) ;

 ^

Spo2_650usec_wemos:44: error: 'ADPS1' was not declared in this scope

 cbi(ADCSRA,ADPS1) ;

            ^

C:\Users\tanze\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:184:25: note: in definition of macro '_BV'

 #define _BV(b) (1UL << (b))

                         ^

C:\Users\tanze\Desktop\Spo2_650usec_wemos\Spo2_650usec_wemos.ino:44:1: note: in expansion of macro 'cbi'

 cbi(ADCSRA,ADPS1) ;

 ^

Spo2_650usec_wemos:45: error: 'ADPS0' was not declared in this scope

 cbi(ADCSRA,ADPS0) ;

            ^

C:\Users\tanze\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:184:25: note: in definition of macro '_BV'

 #define _BV(b) (1UL << (b))

                         ^

C:\Users\tanze\Desktop\Spo2_650usec_wemos\Spo2_650usec_wemos.ino:45:1: note: in expansion of macro 'cbi'

 cbi(ADCSRA,ADPS0) ;

```
Hello,
I am trying to use faster analogRead function. So I want to change prescaler of the ADC in ESP8266 actually in Wemos. I need to change prescaler of ADC to 16 because in my application analogRead speed is very important(like 14-17 microseconds max).
```

me-no-dev, 2016-08-29:
You can not change the prescaler in ESP8266. What you can do is get external ADC (on SPI) and use that.

jrobeson, 2016-10-30:
@me-no-dev : this issue should probably be closed then right?

mph070770, 2016-11-10:
Although this post #696 suggests that there were people who had the ADC working faster (although the link to that page is now broken), is it generally agreed that the ADC is slow? In my tests I can't get it running on a timer any faster than 5KHz.

devyte, 2017-10-13:
Duplicate of #696, will track there. Closing.

closed 2017-10-13 by devyte