esp8266/Arduino #2460: ESP 82661 12F inconsistent with HC-SR04 ultrosonic sensor

Shelva, 2016-08-30:
I am using the above components. But the reading i get from ESP is inconsistent. Is there any extra reset/setting that i must do to use these components together?

It works perfectly on an Arduino Uno board without any change to code.

WereCatf, 2016-08-30:
Are you powering the HC-SR04 from 3.3V or from 5V? It needs 5V for proper operation.

consolacion, 2016-11-22:
Seems like a solved issue but let me just add that I ran in the same "problem" after being led to believe that 3.3 was OK.
I now feed it with 5 Volt and use a level converter. Supposedly the level converter is not necessary on the trigger-pin, only on the echo-pin, but I did on both.
level converter on the echo pin can be as easy as a voltage divider

WereCatf, 2016-11-22:
@consolacion The ESP8266's pins are 5V-tolerant, no level-converter needed -- you can't power it from 5V, it needs to be powered from 3.3V, but the pins themselves don't mind.

consolacion, 2016-11-22:
Thanks for the info. I have done that once with an LCD... didnt work, so ever since I am just putting a voltage converter in. Still good to know they are 5 V tolerant :-)

duncan-a, 2016-11-22:
Unless your LCD was sending data *to* the ESP, what you were seeing was
that your LCD wasn't happy with the 3.3V logic signals *from* the ESP...

That is a totally different thing to the ESP pins being 5V tolerant...

consolacion, 2016-11-22:
I know the difference. I was talking about connecting 5 Volt to 3.3V logic in general. The SR04 is actually sending and receiving to/from the microcontroller

devyte, 2017-10-13:
Not the right place to ask for assistance. Closing per #3655 .

closed 2017-10-13 by devyte