esp8266/Arduino #2461: ESP-12e Exceptions

rudacs, 2016-08-30:
Hardware: ESP-12E
Core Version: 2.3.0

Problem description:
I'm having trouble with my module.
The error does not occur when you start the module, but after some time.

Module:  Generic ESP8266 Module
Flash Size: 4MB (1MB SPIFFS)
CPU Frequency: 80Mhz
Flash Mode: DIO
Flash Frequency: 40Mhz
Upload Using: SERIAL
Reset Method: ck

### Sketch

```#include <ESP8266WiFi.h>
#include <PubSubClient.h>

#define TOPICO_SUBSCRIBE "Envia"
#define TOPICO_PUBLISH   "Recebe"
#define ID_MQTT  "HomeAut"                                
 
//defines - mapeamento de pinos do NodeMCU
#define D0    16
#define LED    0

const char* SSID = "...";
const char* PASSWORD = "...";

const char* BROKER_MQTT = "...";
int BROKER_PORT = 1883;

WiFiClient espClient;
PubSubClient MQTT(espClient);
char EstadoSaida = '0';

void initSerial();
void initWiFi();
void initMQTT();
void reconectWiFi(); 
void mqtt_callback(char* topic, byte* payload, unsigned int length);
void VerificaConexoesWiFIEMQTT(void);
void InitOutput(void);

void setup() 
{
    //inicializações:
    InitOutput();
    initSerial();
    initWiFi();
    initMQTT();
}

void initSerial() 
{
    Serial.begin(115200);
}

void initWiFi() 
{
    delay(10);
    Serial.println("------Conexao WI-FI------");
    Serial.print("Conectando-se na rede: ");
    Serial.println(SSID);
    Serial.println("Aguarde");
     
    reconectWiFi();
}
[... 382 lines cut ...]
0x401011a6: ppEnqueueRxq at ?? line ?
0x40104bd0: wdt_feed at ?? line ?
0x40101612: pp_post at ?? line ?
0x40104be9: wdt_feed at ?? line ?
0x40104be6: wdt_feed at ?? line ?
0x402260e0: ip4_addr_isbroadcast at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/ipv4/ip_addr.c line 63
0x40224bf9: update_arp_entry at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/netif/etharp.c line 466 (discriminator 1)
0x4010020c: _umm_free at /Users/ruda/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1287
0x4010068c: free at /Users/ruda/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1733
0x4021a291: system_pp_recycle_rx_pkt at ?? line ?
0x40106fc4: vPortFree at /Users/ruda/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/heap.c line 18
0x40222152: pbuf_free at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/pbuf.c line 752
0x40210d63: ieee80211_parse_wmeparams at ?? line ?
0x40222169: pbuf_free at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/pbuf.c line 758
0x40206287: pp_attach at ?? line ?
0x401004d8: malloc at /Users/ruda/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1664
0x4010175b: ppCalFrameTimes at ?? line ?
0x40106fb4: pvPortMalloc at /Users/ruda/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/heap.c line 13
0x402222dd: pbuf_alloc at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/pbuf.c line 388
0x4020c649: ieee80211_decap at ?? line ?
0x402251c4: ethernet_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/netif/etharp.c line 1372
0x40205beb: pp_tx_idle_timeout at ?? line ?
0x40205b5e: pp_tx_idle_timeout at ?? line ?
0x40219c57: ets_snprintf at ?? line ?
```

devyte, 2017-10-13:
@rudacs I don't see anything obvious in your code. Is this issue still valid with latest git?

devyte, 2018-01-08:
Closing due to lack of feedback in over 2 months, and age.

closed 2018-01-08 by devyte