esp8266/Arduino #2462: Arduino on ESP8266 has no interface to permanently disable Wifi / Wifi calibration for low power boot-up (not deep sleep)

2dom, 2016-08-30:
### Basic Infos

The ESP8266 produces rather significant power spikes on boot which, so I read, are due to Wifi RF calibration on start-up. I can force no RF calibration after deep sleep but not on a regular power-on boot. 

I find this rather problematic since I have an applications where I have a power source which provides plenty of power for normal (non Wifi) operation however simply cannot supply the >300mA for booting.

I played with the phy_init_data[108] (disable RF calibration for certain number of times) and phy_init_data[114] (rf_cal_use_flash) values in core_esp8266_phy.c with no success.

WereCatf, 2016-08-31:
I don't have the tools to check power-on power-consumption, but after scouring through the code it looks to my eye like modifying phy_init_data[108] **should** work. I have no idea why it doesn't work for you.

2dom, 2016-09-01:
Hmm ... I am wondering where the information in "phy_init_data" is actually used. It appears to be set when "__wrap_register_chipv6_phy" is called but I cannot find a call anywhere in the Arduino ESP8266 source or the ESP SDK source....

WereCatf, 2016-09-01:
Isn't the ESP SDK partly closed-source? At least I cannot find sources for certain parts of it, including libphy.a

2dom, 2016-09-03:
Ok...I used huge capacitors (2x1000uF) to get the ESP to boot whilst connected to a weak power supply. But this is hardly elegant. The lack of an interface to disable WiFi for power-on boot remains...

devyte, 2017-10-13:
@2dom I believe that there have been changes in rf calibration in recent SDK (I think since SDK2.0.0).
Is this issue still valid with PR #3215 ?

5chufti, 2017-12-28:
seems to be same basic cause as in #3408

devyte, 2018-12-05:
It is now possible to define a void preinit() function that executes early on bootup before C++ Dynamic Init phase. In this function it should be possible to turn off things like rf cal via sdk calls.
Ref: #5395 .
For the deep sleep thing please continue discussion in #3408 .
Closing.

closed 2018-12-05 by devyte