esp8266/Arduino #2464: Asynchronous WPS

marvinroger, 2016-08-31:
The current WPS implementation is synchronous. Given that ESP8266WiFi can run async with some events handler, I would like to think this is also doable with WPS, so WPS configuration can happen in the background.

Thanks for your hard work, once again.

JimmyTheOne, 2017-04-21:
Heey, are there any updates on this? :)

devyte, 2017-10-13:
@marvinroger the SDK implementation is async. The core wrapper method is sync in the sense that it waits for the callback to fire before returning. While it is waiting, control is returned to the SDK layer.
Considering that the WPS configuration is something that is done once, I don't see a need for a full async implementation. The implementation would add complexity to the current code base for insufficient need and use cases.
Also, I'd like to mention here that WPS is considered insecure, and shouldn't be used unless you are absolutely sure that your router has a workaround for the security issue.
If you really need this, you can look at the current implementation in ESP8266WiFiSTA.cpp, and look at how the callback is set. With that, you can roll your own implementation.
Closing.

closed 2017-10-13 by devyte