esp8266/Arduino #2466: I2C SS conflicts with 'gpio15 must be low during boot'

supersjimmie, 2016-08-31:
In common.h the I2C SS is specified as:
```static const uint8_t SS = 15;```
But for a normal boot, gpio15 must be pulled low.

When I attach an I2C device to MOSI, MISO, SCK and SS, how can I be certain the esp will still boot? (in my case it fails 'randomly')
Would a pulldown resistor on gpio15 be enough to fix this correctly? Or should I remove the device' SS at each boot?

me-no-dev, 2016-08-31:
can not use a pulled down pin (GPIO15) for I2C ;) I2C can work only with pull-ups.

me-no-dev, 2016-08-31:
MOSI, MISO, SCK and SS are SPI pins and if you use SS for slave select, you need to make sure that the salve is not pulling it up by pull-up.

supersjimmie, 2016-08-31:
So what you are saying is that I cannot pull-down and I2C works with pull-up, but it should not be pulled-up for normal boot operation.... Now I am confused.  :(

supersjimmie, 2016-08-31:
@me-no-dev the datasheet of my slave device says:
CC1101 is configured via a simple 4-wire SPI compatible interface (SI, SO, SCLK and CSn) where CC1101 is the slave. 
...
The CSn pin must be kept low during transfers on the SPI bus.

So in my code I can say 
```
pinMode(SS, OUTPUT);
digitalWrite(SS, LOW); // only set to HIGH when needed
```
But then still, if you are saying I need pull-up to make I2C work, how can I avoid a wrong boot-mode?
Both the esp and the slave are 'undefined' high or low if not pulled-up/down?

duncan-a, 2016-08-31:
The CC1101 is an SPI-bus device - it does not use the I2C-bus...

supersjimmie, 2016-08-31:
@duncan-a / @me-no-dev   I know, I changed my question because I ment SPI instead of I2C.

me-no-dev, 2016-08-31:
having a pull-down on pin15 does not break SPI communication. You can in your code init the pin HIGH because that will happen after boot anyway.

supersjimmie, 2016-08-31:
Thanks @me-no-dev if that is possible then that is how I will start building.

closed 2016-08-31 by supersjimmie

Maigre, 2017-09-18:
Hi, sorry for the bump, 
but i am facing the same issue here: 
i am plugging a wiz5500 to esp8266 via SPI,
and the ESP8266 fails to boot as soon as i plug the SS pin..
How do you handle that ? Should i wire a resistor from gpio15 to GND ?
SPI will still be working after that ?
Thanks !

vicnevicne, 2017-09-18:
Hi,
In my case, a pull down was not the solution because the master actively drove the SS up.
My solution was to use another pin as an "SS enable" (SS_EN), with 3 resistors and a transistor. On boot-up, SS_EN is in high impedance and guarantees that GPIO15 is kept down, but as soon as the setup() drives SS_EN high, the master SS signal is passed on to GPIO15. 
The schematic is as follows: http://blog.deconinck.info/public/WAHOOcorder/.2017-01-18_2040_m.jpg
Probably a simpler circuit to do the job, but that's what I came up with and it works... feel free to comment or propose alternatives of course...

Maigre, 2017-09-20:
Hi !
i end up wiring a 4.7k resistor between GPIO15 (D8) and GND.
The GPIO15 is used as SS without any trouble, and it boot correctly now.
Thanks for your help and detailed answer ! It pointed me in the good direction :)
Best,
Thomas

shivapower1985, 2021-06-15:
> Hi, sorry for the bump,
> but i am facing the same issue here:
> i am plugging a wiz5500 to esp8266 via SPI,
> and the ESP8266 fails to boot as soon as i plug the SS pin..
> How do you handle that ? Should i wire a resistor from gpio15 to GND ?
> SPI will still be working after that ?
> Thanks !

Hi Maigre,
First of all thanks for your suggestion to wire a resistor between GPIO15 and GND, and after hours of search I am able to upload my sketch with my CS pin connected at GPIO 15. 
I want your help in connecting W5500 lite with ESP8266. I find difficulty in configuring DHCP. When I upload the sketch for setting the IP it throws "IP unset" and returns "0" for "Ethernet.begin(mac)". 
my pin connections are,
W5500 lite   -    ESP8266
---------------------------
        V          -     3.3v
        G          -     GND
       SCK       -     D5
       MI         -     D6
       MO       -     D7
       CS         -     D8

Sketch
#include <SPI.h>
#include <Ethernet.h>

byte mac[] = {  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
//IPAddress ip(192,168,2,2);

void setup() {
  Serial.begin(9600);
  
  Ethernet.init(15);
  pinMode(15, OUTPUT);
  digitalWrite(15, LOW); // only set to HIGH when needed

  Serial.print(Ethernet.begin(mac));

  Ethernet.begin(mac);

  Serial.println(Ethernet.localIP());
}

void loop() {
}

My OUTPUT:
0 
(IP unset)

shivapower1985, 2021-06-15:
Hi Maigre,
First of all thanks for your suggestion to wire a resistor between GPIO15 and GND, and after hours of search I am able to upload my sketch with my CS pin connected at GPIO 15. 
I want your help in connecting W5500 lite with ESP8266. I find difficulty in configuring DHCP. When I upload the sketch for setting the IP it throws "IP unset" and returns "0" for "Ethernet.begin(mac)". 
my pin connections are,
W5500 lite   -    ESP8266
---------------------------
        V          -     3.3v
        G          -     GND
       SCK       -     D5
       MI         -     D6
       MO       -     D7
       CS         -     D8

Sketch
#include <SPI.h>
#include <Ethernet.h>

byte mac[] = {  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
//IPAddress ip(192,168,2,2);

void setup() {
  Serial.begin(9600);
  
  Ethernet.init(15);
  pinMode(15, OUTPUT);
  digitalWrite(15, LOW); // only set to HIGH when needed

  Serial.print(Ethernet.begin(mac));

  Ethernet.begin(mac);

  Serial.println(Ethernet.localIP());
}

void loop() {
}

My OUTPUT:
0 
(IP unset)