esp8266/Arduino #2467: Use Arduino Mega + ESP8266

LuizHAP, 2016-08-31:
Hi guys, i need some help in my Project.
This library works for connection ESP8266 with Arduino Mega like a connection below?

ESP8266 TX -> Arduino MEGA RX (19)
ESP8266 RX -> Arduino MEGA TX (18) with 3.3v

I will use this project for my Home Automation, then will receive HTTPS request from a Node.js.

Can you help me?

Thanks!

supersjimmie, 2016-09-03:
Take a look at https://github.com/sparkfun/SparkFun_ESP8266_AT_Arduino_Library
(don't let the picture mislead you, just look at the code, I think that is what you need)

Bushii7, 2017-07-26:
i am also having difficulties in connection with arduino mega 
can u guys help me with the connections?PIN config ?

devyte, 2017-10-13:
Not the right place to request help for your particular project. Closing per #3655 .

closed 2017-10-13 by devyte