esp8266/Arduino #2468: WiFiMulti.run looped when passphrase is incorrect

denisn73, 2016-09-01:
Hi. I'm use WiFiMulti.run in AP_STA mode, and when i'm change passphrase of remote AP, programm loops in WiFi multi.run. I assume that the program is looped here:
---------------------------------
// wait for connection or fail
while(status != WL_CONNECTED && status != WL_NO_SSID_AVAIL && status != WL_CONNECT_FAILED) {
  delay(10);
  status = WiFi.status();
}
---------------------------------
Because of that I can not go to the server to reconfigure the passphrase, as 'server.handleClient();' function in the my main loop is not processed.
How can I stop WiFi Multi if the key phrase has changed and accordingly become incorrect?
Please fix this bug.

denisn73, 2016-09-01:
I'm correct function run() to:
----
// wait for connection or fail
//while(status != WL_CONNECTED && status != WL_NO_SSID_AVAIL && status != WL_CONNECT_FAILED) {
while(status != WL_CONNECTED && status != WL_NO_SSID_AVAIL && status != WL_CONNECT_FAILED && status != WL_DISCONNECTED) {
  delay(10);
  status = WiFi.status();
 }

---
And now this work for me.

devyte, 2017-10-13:
@denisn73 is this issue still valid in latest git? does it also happen with PR #3215 ?

devyte, 2018-01-12:
Closing as duplicate of #3993 .

closed 2018-01-12 by devyte

ConstantinCreations, 2022-08-23:
Still an issue. Wrong password and wifimulti will fail to try another network.