esp8266/Arduino #2469: GPIOs to hold state during Deep Sleep

vlast3k, 2016-09-01:
### Basic Infos

#### Hardware
Hardware:	        ESP-12E
Core Version:      	latest github

### Description

I would like to setup the ESP8266 in a way that the GPIOs hold their state while in deep sleep. By default it looks like when the device goes into deep sleep or wakes up, GPIOs that have been set to HIGH, go LOW (which causes an external sensor that is connected to restart, and then it needs 2 minutes to warm up again).
Is there a way to set them up in a way that they are not switched off ?

me-no-dev, 2016-09-01:
No. When in deep sleep, the GPIO hardware is turned off. Only the RTC part is on and counting :)

vlast3k, 2016-09-01:
ahhh
thanks :)

closed 2016-09-01 by vlast3k

pieman64, 2016-09-01:
@vlast3k you can store the pin state in EEPROM or with something like a Smartphone app such as Blynk and restore after deepsleep but it wouldn't stop the pin going low during sleep. All it would do is ensure it goes back high after sleep. This works for some applications but not many.

vlast3k, 2016-09-02:
after reading a bit more in the recent documentation (seems like it was updated few weeks ago), it seems **like in deepSleep the GPIOs maintain their state, yet the drive strength is 2uA**
http://espressif.com/sites/default/files/documentation/9b-esp8266_low_power_solutions_en.pdf
Why it didn't work in my case is because i did not realize that i had connected ti to GPIO15, which i anyway have pulled down by a 10k resistor :(

rumpeltux, 2019-09-05:
For everyone else coming here searching about this issue:
[the updated datasheet (v1.5)](https://www.espressif.com/sites/default/files/documentation/9b-esp8266-low_power_solutions__en.pdf) (table 1-1) now officially mentions that GPIO state in deep sleep is undefined, i.e. **GPIOs don’t maintain their state**. (The 2μA number still seems wrong from my measurements).

ChumKiu, 2019-11-05:
> For everyone else coming here searching about this issue:
> [the updated datasheet (v1.5)](https://www.espressif.com/sites/default/files/documentation/9b-esp8266-low_power_solutions__en.pdf) (table [...]

Hi, referring same datasheet (V1.5 - 2019) at page 5/9 is clearly stated:

"In Deep-sleep mode, GPIO maintains its level state which has a 2-μA drive strength."

so, if this is correct, the state is retained **only if the output is not loaded over the specified current**, this means then that a load with a impedance greater than 1.65 Mohm shall be used otherwise the readed value will be not the expected one.

Make sense?

rumpeltux, 2019-11-05:
> so, if this is correct, the state is retained **only if the output is not loaded over the specified current**, this means then that a load with a impedance greater than 1.65 Mohm shall be used otherwise the readed value will be not the expected one.
> 
> Make sense?

This doesn’t match my observations (I don’t recall exactly, but the GPIOs were driven in the opposite direction of what I set them too at >2μA). I believe the paragraph is a wrong left-over from previous versions of the document.

marno11, 2019-12-17:
Maybe this is a silly suggestion: But if you know that you always want a pin to be high during sleep could you use GPIO0 or GPIO2 (which have pull-ups), or if you know you always want the pin to be low during sleep use GPIO15 (which has a pull-down)?

As for maintaining an arbitrary pin state during deep sleep, that appears to not be practically possible unless you add some extra hardware specifically to do this.

Tech-TX, 2020-11-27:
I realize this is an old, closed issue, but here's my recent observations since Google is still finding this for info.

Marno11's suggestion won't work, as GPIO0 and GPIO2 both wiggle during the boot code.  A relay would chatter.  GPIO15 has a strange 0.7V level on it that may cause a buffer to see a HIGH during the boot code.

After a Deep Sleep Wake, _some_ of the pins will hold their previous value, but it appears to be the 2uA weak pull-up or pull-down that was activated when it went to Deep Sleep.  Most pins don't return to active OUTPUT but will hold the levels below:

GPIO4, GPIO5 will maintain a weak LOW after Deep Sleep.
GPIO4, GPIO5, GPIO12, GPIO13, GPIO14 will maintain a weak HIGH after Deep Sleep.
GPIO3 appears to maintain an active HIGH after Deep Sleep; I get a strange voltage level on my board.

You'd need to buffer those pin levels with a FET or other high impedance buffer since they're in 2uA drive; an NPN or PNP transistor would pull too much current and lose the level.

Here's the little chunk of test code I used.  It only sets the **OUTPUT, pinLevel** after power-up reset or external reset; the pins aren't touched after Deep Sleep Wake.
```cpp
#include <ESP8266WiFi.h>
uint8_t pinLevel = 1;  // change to 0 to check output LOW

void setup() {
  String resetCause = ESP.getResetReason();
  if (resetCause != "Deep-Sleep Wake") {
    pinMode(0, OUTPUT);
    digitalWrite(0, pinLevel);
    pinMode(1, OUTPUT);
    digitalWrite(1, pinLevel);
    pinMode(2, OUTPUT);
    digitalWrite(2, pinLevel);
    pinMode(3, OUTPUT);
    digitalWrite(3, pinLevel);
    pinMode(4, OUTPUT);
    digitalWrite(4, pinLevel);
    pinMode(5, OUTPUT);
    digitalWrite(5, pinLevel);
    pinMode(12, OUTPUT);
    digitalWrite(12, pinLevel);
    pinMode(13, OUTPUT);
    digitalWrite(13, pinLevel);
    pinMode(14, OUTPUT);
    digitalWrite(14, pinLevel);
    pinMode(15, OUTPUT);
    digitalWrite(15, pinLevel);
  }
}

void loop() {
  delay(5e3);
  ESP.deepSleep(5e6, WAKE_NO_RFCAL);
}
```