esp8266/Arduino #2471: OTA serial port no longer appears in arduino IDE - restart the system, works again

odilonafonso, 2016-09-01:
### Basic Infos

#### Hardware
Hardware:		ESP-12 Nodemcu V2
Core Version:      	2.3

### Description
OTA was working perfectly well and a sudden, for no apparent reason, stopped working.
Now I compile the sketch on the serial port of the Arduino IDE and then the serial port OTA is not offered for use.
I've tried resetting the module, closed and opened the IDE many times. Nothing solves.
The OTA port no longer appears in the IDE to be used as uploading port.
I decided restart the notebook; started working again.
Please, anyone have any idea what might have happened?

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12e module)
Flash Size: 		4MB/3MB
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	?40Mhz?
Upload Using:		OTA / SERIAL
Reset Method:		nodemcu

### Sketch

Not relevant (suppose)
Both my sketch as BasicOTA from samples have the same problem

SrikanthBala, 2016-09-07:
I had a similar problem and struggled with it for a while. I later found that the reason is that for some reason the Bonjour Services sometimes shows as stopped under the running 'Services' on my Windows 10 machine. On rebooting, the Bonjour Service starts (as during installation is set to autostart on reboot,) and everything works fine.  
Next time the OTA port disappears it may be worthwhile to check under the Services tab of Windows to see if Bonjour services are running.
Why does it stop ? I have no idea.

odilonafonso, 2016-09-08:
Hi @SrikanthBala 
Thanks for your help. But this reason is not applicable to me - I use Linux.
May be can be useful for someone else.
Anyway, when it happens I do a restart on Linux and back to work.

odilonafonso, 2016-10-03:
I can not figure out what's going on. I have 3 devices connected, recorded with OTA. Two NodeMCU and one Wemos.

For several days, I could see and record through the serial ports that OTA offered me;

Now, for no apparent reason, the Arduino IDE does not show me any of the three doors.
Have opened and closed the IDE, I reseted the notebook. Nothing solves.

Any idea what might be happening?

Grateful.

pieman64, 2016-10-03:
@odilonafonso same for me on Windows 8.1 even after a laptop restart.

I am on 1.6.12, you?

Could be Bonjour related for me and I will check this but perhaps a 1.6.12 bug?

odilonafonso, 2016-10-03:
@pieman64 I am on 1.6.10.

My MDNS is mdns linux avahi

I think is not a MDNS problem

I have another mdns stations - local host, router. And all f them works ok.

Right now OTA is working. But I was a few hours without.

Patience, apparently, OTA works when it wants. Or until the day I figure out what I'm doing wrong.

closed 2016-10-21 by odilonafonso