esp8266/Arduino #2473: Receiving too much UDP packets makes it impossible to send any packets

lewan0996, 2016-09-02:
Hi,

I am developing an android app which sends microphone data to the ESP via UDP. ESP records data from SPI ADC and sends them via UDP back to android. The problem is that if I increase the recording sample rate in android above 24khz, ESP won't send packets any more(they doesn't arrive). It seems that if I overflow the UDP rx buffer, it is unable to send data. Is there any solution to this problem?

I send packets like that:
```cpp
    uint16_t sample = adc_->readADC(0);	
	lowBytes = sample & 0xff;
	highBytes = sample >> 8;

	buffer_[bufferIndex] = lowBytes;
	buffer_[bufferIndex + 1] = highBytes;
    beginPacket(address_, port_);
	write(buffer_, length);	
	endPacket();
```
And receive like that:

```cpp
    size = parsePacket();	
	read(buffer_, Config::bufferSize);
```

Packet size is 1024 right now.

devyte, 2017-10-13:
@lewan0996 Is this issue still valid? Did you try a yield() after endPacket?
Please try PR #3362 .

lewan0996, 2017-11-07:
Hi, thanks for the reply. Unfortunately it was so much time ago and we are no longer into this project. As far as I remember putting yield anywhere didn't change anything.

closed 2017-11-07 by lewan0996