esp8266/Arduino #2478: OTA often fails with watchdog timer reset

jackjansen, 2016-09-03:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.3.0

### Description

OTA from the Arduino IDE often fails for me (some days 80% of the time). If I have the serial port connected (just for debug output, not for uploading) what I see is that the watchdog timer triggers, the 8266 resets, and then goes off to never-never land. Power cycling is required to get it working again.

I'm using ArduinoOTA, not the http server. I've added wdtFeed() calls to the start/progress/end callbacks, which has helped a bit, but I'm still getting the resets fairly often. It may happen more of the WiFi is busy.

Could it be that there's a wdtFeed() or similar call missing in the ArduinoOTA inner loop?

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Upload Using:		OTA

penfold42, 2016-09-04:
Have you been resetting the ESP after a serial upload ?

This is required before using OTA to update it or else it will crash.

jackjansen, 2016-09-04:
Hmm, I don't know... Whether or not I did a reset may be the difference between sometimes failing, sometimes not. I will experiment and report back here.

penfold42, 2016-09-04:
Yup - it was driving me nuts until I stumbled across it in another issue

namirda, 2016-10-05:
Hi Jack - you never came back to say if you had solved your problem. I would very much like to know.

I have exactly the same issue - ArduinoOTA fails most of the time with a wdt reset. 

I dug into the ArduinoOTA library a little and found that the reset was occurring in the call to WiFiUDP::stopAll() in method ArduinoOTAClass::_runUpdate(). 

Adding a yield() before the call to WiFiUDP::stopAll() did not help and so I simply removed it - I do not have any instances of WiFiUDP in my sketch.  All my problems are now gone and OTA appears to work perfectly!

Maybe this helps you.

jackjansen, 2016-10-06:
Sorry for not reporting back! The comment by @penfold42 hit the nail on the head: if you have uploaded via serial you **must** issue a reset before attempting an OTA upload. So I just made that reset part of my workflow while debugging my OTA code.

It would be nice if this was documented somewhere (if not fixed).

penfold42, 2016-10-06:
IIRC it can't be fixed

Maybe a reminder from the ESP upload tool would be nice

devyte, 2017-10-13:
After a serial upload, the ESP must be manually reset before an OTA can work (actually, I think before a software reset as well). And I also think it can't be fixed.
Closing.

closed 2017-10-13 by devyte