esp8266/Arduino #2479: There is a constant indicating the board being compiled ?

odilonafonso, 2016-09-03:
### Basic Infos

#### Hardware
Hardware:		ESP-12 NodeMCU or Wemos D1 Mini
Core Version:      	2.3

### Description

There is some constant that I can use to identify the board being compiled?

For example, I use a same sketch to compile both a ESP-12e as one WEMOS Mini D1 or even an ESP-01.
Today manually I have to do the following:
(for example)

```
#define ESP-01
// #define ESP-12
// #define WEMOS

#if defined ESP-01
#define STATUS_LED 0
#endif

#if defined ESP-12
#define STATUS_LED 14
#endif

#if defined WEMOS
#define STATUS_LED 5
#endif

```

But it would be interesting to use a constant that already indicate the board being used. Suppose the build process when identifying the chosen board in the Arduino IDE, replicate this choice by setting a constant.

### Settings in IDE

Module:  			ESP-12 NodeMCU or Wemos D1 Mini
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		dio
Flash Frequency:	?40Mhz?
Upload Using:		OTA / SERIAL
Reset Method:		nodemcu

### Sketch
Not relevant

pieman64, 2016-09-03:
Yes board types are available as they were for Arduinos.

pieman64, 2016-09-03:
I don't think there is a difference between a 1 and 12 within the IDE even though they have very different pin quantities. We cheat for ESP12's and use NodeMCU so for you it would be something like this:
```
  #ifdef  ARDUINO_ESP8266_WEMOS_D1MINI  // WeMos mini and D1 R2
    #define STATUS_LED 5
  #elif ARDUINO_ESP8266_ESP01           // Generic ESP's use for 01's
    #define STATUS_LED 0    
  #elif ARDUINO_ESP8266_NODEMCU               // Wio Link and NodeMCU 1.0 (also 0.9), use for ESP12
    #define STATUS_LED 14       
  #endif 
```

odilonafonso, 2016-09-03:
Hi @pieman64.
Grateful for your help.
Please, where did you find this information? These constants?

pieman64, 2016-09-03:
@odilonafonso for me they are contained in:

C:\Users\maja\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\boards.txt

odilonafonso, 2016-09-03:
This constants are not defined for me. I will give a look on my boards.txt.
Thanks !

odilonafonso, 2016-09-03:
I have this in my boards.txt:

`nodemcuv2.build.board=ESP8266_NODEMCU`

but the constant is not defined....

[boards.txt](https://github.com/esp8266/Arduino/files/453647/boards.txt)

WereCatf, 2016-09-04:
@odilonafonso It's defined as ARDUINO_ESP8266_NODEMCU then.

All the different boards follow the same scheme: ARDUINO_{build.board}, not just {build.board}, that's why it becomes ARDUINO_ESP8266_NODEMCU in your case.

odilonafonso, 2016-09-04:
Ahhhhhhh !!! Now works !!
Thanks @WereCatf .
Thanks also to you, @pieman64. I had not understood right.

closed 2016-09-04 by odilonafonso

enlinea777, 2018-08-17:
hello.
how could I extract or the other parameters for example:
name, CpuFrequency, mcu, flash_mode, flash_freq?

JJiri, 2018-09-16:
any news?
for last question: 
I need info for FALSH memory...