esp8266/Arduino #2480: Manually specify OTA ports in arduino Ide ?

penfold42, 2016-09-04:
My laptop has a firewall I can't administer and it blocks UDP 5353 inbound.
This stops AirTunes but also stops the OTA advertisements making it it.

How can I manually fake the existence of the ESP devices so they show up in the ide ?

Thanks !

lucky62, 2016-12-03:
+1 vote to add OTA port(s) into arduino IDE manually.

vic10us, 2016-12-11:
+1 vote

devyte, 2017-10-13:
@penfold42 is this issue still valid? I understand that ArduinoOTA advertises using mDNS, and that you can specify which port to use. This is done in your code.

penfold42, 2017-10-13:
Yes, still an issue.

A way to fake / manually add the entries would be great

igrr, 2017-10-14:
I guess this is more of an IDE issue than ESP8266 then? I.e. it applies to
any board which uses mDNS for OTA, not just the ESP8266.

penfold42, 2017-10-14:
Yes - I didn't realise the OTA subsystem is a base arduino IDE capability.

atrent, 2017-12-07:
Actually I don't think opening the 5353 port does solve the issue (I've tried adding a rule and it still fails the uploading process).

I tried using the CLI command:
python .arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/espota.py -i HOSTIPADDR -p HOSTPORT -I REMOTEIPADDR -P REMOTEPORT -f SKETCH.bin 
and it JUST works (by adding a rule for HOSTPORT)

It would be great to be able to specify one port to listen onto, maybe in a config file

atrent, 2017-12-07:
SOLVED!

In file ".arduino15/packages/esp8266/hardware/esp8266/2.3.0/platform.txt"

change line:
tools.esptool.upload.network_pattern="{network_cmd}" "{runtime.platform.path}/tools/espota.py" -i "{serial.port}" -p "{network.port}" "--auth={network.password}" -f "{build.path}/{build.project_name}.bin"

to:
tools.esptool.upload.network_pattern="{network_cmd}" "{runtime.platform.path}/tools/espota.py" -i "{serial.port}" -p "{network.port}" -P 8266 "--auth={network.password}" -f "{build.path}/{build.project_name}.bin"

this way the LISTENING port on your development pc is fixed to 8266 (of course you can change it) and you can write a rule on your firewall

atrent, 2017-12-07:
better yet!

in "preferences.txt" add

ota.hostport=8266

and in file ".arduino15/packages/esp8266/hardware/esp8266/2.3.0/platform.txt" change the line in:

tools.esptool.upload.network_pattern="{network_cmd}" "{runtime.platform.path}/tools/espota.py" -i "{serial.port}" -p "{network.port}" -P "{ota.hostport}" "--auth={network.password}" -f "{build.path}/{build.project_name}.bin"

herrold, 2017-12-07:
@devyte 

he said, earlier that 5353 was being blocked

[herrold@centos-7 ~]$ grep 5353 /etc/services 
mdns            5353/tcp                # Multicast DNS
mdns            5353/udp                # Multicast DNS

actually I would make the default in @atrent 's proposal, the IANA assigned $DEFAULT of 5353, rather than a local side port

herrold, 2017-12-07:
@penfold42 

This issue is in a closed state -- could you please re-open it so it does not get forgotten, and gets documented

atrent, 2017-12-07:
of course, you can change the config to any port you want :)

atrent, 2017-12-07:
actually, let me say that 5353 should NOT be used since it's dedicated to mdns service, while the upload of the new sketch via network is another service altogether, i.e. we should use an "unused" port

mdns is a discovery service, it's needed to find the device, the following communication is another story

herrold, 2017-12-07:
@atrent 

understood, but there is 'legacy' backward compatibility to match existing documentation -- a clean break at a major release epoch makes sense, though

I'd be thrilled on a small IoT friendly avahi daemon and hand-off for OTA purposes.  I have not looked -- do you know of a less 'weighty' suite (the current Avahi tarball is 1.2 meg big)

devyte, 2017-12-07:
@igrr what do you think of @atrent 's proposal to change platform.txt?

penfold42, 2017-12-07:
The subject could be misleading. 
Ports on my case refers to the list of ports in the arduino GUI where the OTA targets show up (next to COM3, COM17 etc)

This relies on mDNS advertisements making it past the firewall on port 5353.
As it is locked down, I was looking for a way to fake the entries in the GUI.

atrent, 2018-02-14:
Sorry for "bumping", did you think about my proposal?

veleek, 2019-03-05:
@atrent, there's a slight misunderstanding here I think.  @penfold42 was looking for a way to manually specify a TARGET destination for an OTA upload in the Arduino GUI.  As a good example of this, on Windows machines, it can be difficult to get mDNS working properly which means that even though I have an ESP OTA service running (I can see it using my mDNS browser on my iPhone) I'm unable to see the "Network Port" in the Arduino GUI. 

However, I can still manually connect to the ESP using it's IP address, and since the IP doesn't change very often on my home network it would be very helpful to be able to manually setup a network port in the Arduino UI in order to be able to deploy an OTA update to my ESP from the UI.

veleek, 2019-03-05:
For anybody interested in a useful workaround, if you're using VSCode you can create a [custom task](https://code.visualstudio.com/docs/editor/tasks#_custom-tasks) to execute the `espota.py` directly and specify the IP and destination port.  It's a quick, easy, reusable way of deploying OTA updates.

Example `tasks.json`:
```jsonc
{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "label": "OTA Upload",
            "type": "shell",
            "command": "py",
            "args": [
                "%LOCALAPPDATA%\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\tools\\espota.py",
                "-i",
                "192.168.29.243",
                "-p",
                "8266",
                "-f",
                "${workspaceFolder}\\..\\build\\${fileBasename}.bin",
            ],
            "options": {
                "shell": {
                    "executable": "cmd.exe",
                    "args": [ "/c" ]
                }
            }
        }
    ]
}
```

This task assumes that you have specified that the build output go into `..\build` and you'll need to update the location for `espota.py` if you're using a different version.  Also by default it will look for a file `.bin` file with the same name as the `.ino` sketch you have open (e.g. `sample.ino.bin`).

colinl, 2019-10-09:
Is there still no solution to this?  The network port for OTA only rarely appears in Tools > Ports so I cannot upload OTA.  I am running on Windows 10.  If I could specify the device IP address manually then that would (I think) solve my problem.

d-a-v, 2019-10-09:
You can
* run a terminal,
* locate the build directory where you can find `sketch.bin` freshly compiled,
* get into the core directory (`cd \path\to\arduino15\hardware\...\esp8266\tools`) where `espota.py` is,
* run it:
```
python3/python3.exe espota.py -p <ip-address> -f c:\path\to\sketch.bin
```
* write the documentation
* post it in a PR

colinl, 2019-10-09:
Thanks.  I looked for that script earlier but couldn't find it.  Will have another look.

MaBi6411, 2020-05-02:
@d-a-v
> ```
> python3/python3.exe espota.py -p <ip-address> -f c:\path\to\sketch.bin
> ```

argument for *< ip-address >* is **-i**
<br><br>
I created a very easy tool that do the same thing with a GUI.<br>
You can find it [here](https://github.com/MaBi6411/arduino-esp8266-espotaLauncher).

d-a-v, 2020-05-02:
@MaBi6411 Thanks I fixed my comment.
Your tool is interesting, but it needs an external compiler and may be available only on windows.
You may make a pull request to mention in documentation,
You could also write one in python3 using [tkinter](https://docs.python.org/3.3/library/tkinter.html) as gui as it seems to me it is the most cross-platform and simple way to get a gui not requiring any further tool/compiler.

jandrassy, 2020-05-02:
if espota were installed as an IDE tool, one could configure it as programmer (and it would be part of referred packages problems solution too). examples for the arduinoOTA tool:

programmer.txt:
```
arduinoOTA104.name=Arduino OTA (192.168.1.104)
arduinoOTA104.program.tool=arduinoOTA
arduinoOTA104.ip=192.168.1.104
```

platform.local.txt
```
## arduinoOTA as programmer. add entries with {ip} into programmers.txt
tools.arduinoOTA.cmd={runtime.tools.arduinoOTA.path}/bin/arduinoOTA
tools.arduinoOTA.program.params.verbose=
tools.arduinoOTA.program.params.quiet=
tools.arduinoOTA.program.pattern="{cmd}" -address {ip} -port 65280 -username arduino -password password -sketch "{build.path}/{build.project_name}.bin" -upload /sketch -b
```

MaBi6411, 2020-05-04:
@d-a-v 
A program that makes the same compile functions of _Arduino IDE_ except editing it's not easy for me... and I'm totally new with python.
I think it's more easy to change the official IDE with possibility to insert _IP_ and _password_ of _OTA target_ ...My solution is temporary

MaBi6411, 2020-05-04:
@d-a-v 
> You may make a pull request to mention in documentation<br>

Where do I have to do it? Could you explain to me please? I'm totally new in GitHub.

SheetLightning, 2020-06-12:
Just want to add another vote for manual specification of an OTA target. I have found the automated method to be hit-and-miss when it comes to showing the target in the target ports list.

SheetLightning, 2020-06-12:
I would like to add another vote in favour of being able to manually specify the OTA target. Like others I have found the automated solution to be rather hit-and-miss when it comes to detecting and showing the OTA target. If I am to use this seriously then I need a reliable method of OTA updating my devices.

koshkamau, 2021-08-12:
I'm voting for manual specification of an OTA target too.
OTA is completely unreliable for me (Win10). Right now I was forced to reboot my laptop to see OTA targets in IDE.
Disabling firewall, reconnecting to WiFi and restarting IDE didn't help.

TeguhTeknisi, 2022-08-26:
I'm voting for manual specification of an OTA target too.

JAndrassy, 2022-08-26:
https://github.com/esp8266/Arduino/pull/8432

Vlad-Star, 2025-04-30:
I can' t believe, guys... this issue was opened in 2016 and still has not been resolved by 2025... :( 

My development board is in a separate LAN from the computer with IDE, so I know it won't be "magically" discovered via mDNS. I also know the IP/port where it's listening at, and that my firewall is configured to allow necessary communication between the networks. I am on IDE 2.3.6/Win11 and don't see any way to add the OTA IP/port manually.

Please, implement.