esp8266/Arduino #2481: SDWebServer printDirectory CONTENT_LENGTH_UNKNOWN doesn't send last-chunk

JamesNewton, 2016-09-04:
### Description
In the sample code, in the printDirectory routine, which is one of the only examples I can find where anyone uses 
server.setContentLength(CONTENT_LENGTH_UNKNOWN);
what closes the connection after the listing is returned? 

Shouldn't there be a server.sendContent(""") at the end of the procedure?

When chunking is used, and no Content-Length is supplied in the Header of an HTTP 1.1 response, the last chunk is supposed to have zero length...
http://greenbytes.de/tech/webdav/rfc2616.html#rfc.section.3.6.1
http://greenbytes.de/tech/webdav/rfc2616.html#introduction.of.transfer-encoding

#### Hardware
Hardware:			ESP-12

### Settings in IDE
Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu... actually manual.

### Sketch
/libraries/ESP8266WebServer/examples/SDWebServer/SDWebServer.ino

### Debug Messages
None... dang it... I don't know how to spy on the HTTP data.

What bothers me most is that the code appears to work... and I can't figure out why... Are modern browsers just closing the connection after a while? Nothing in the spec about that which I can find...

bbx10, 2017-07-24:
@JamesNewton I confirmed the same problem. Chromium gets stuck waiting for more data without the 0 length chunk. For example, `http://<ESP IP addr>/list?dir=/` works correctly only after the 0 length chunk is added. This bug affects edit mode as well. The left side panel is supposed to show the filenames but is blank without this fix. I suggest submitting a PR. I will fix the ESP32 version.

devyte, 2017-10-13:
The example still doesn't have the empty chunk in latest git.

PR #6808 (2019-11-19): Terminate the chunked printDirectory in SDWebServer
Fixes #2481

Send a 0-len chunk when the directory is completed (in a chunked HTTP
transfer) to terminate the HTTP transfer properly.
closed 2019-11-19 by d-a-v