esp8266/Arduino #2482: How to know which serial port OTA uses to upload the sketch

odilonafonso, 2016-09-04:
### Basic Infos

#### Hardware
Hardware:		ESP-12
Core Version:      2.3.0-rc2

### Description

I do not know how to configure Putty (or other serial monitor) to monitor messages from OTA to upload the sketch.
The OTA process is working very well, but I would like to see the messages that my sketch is sending through the serial. It's possible ?

### Settings in IDE

Module:  			ESP-12 NodeMCU 1.0
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Upload Using:		OTA

krzychb, 2016-09-04:
Hi @odilonafonso,

It is possible and also recommended for troubleshooting. Please check https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#troubleshooting

To know the serial port, I would:

- disconnect module from the PC,
- check the list of USB devices visible by O/S
- connect module back
- check what new port has been detected by the OS.

Krzysztof

supersjimmie, 2016-09-04:
I would say the same port that you have ever used to upload the sketch before you used OTA...

odilonafonso, 2016-09-05:
Thanks @krzychb and @supersjimmie for help.

I had read this documentation. Sometimes.... :( 
I tried to use the port that the IDE was using before attempting to use OTA to upload.

I'm right now in the ESP8266 powered only and running without USB connection to my laptop.
When researching the ports available for access by the Arduino IDE, I can see OTA port:

**`Port: "esp8266-179eaa on 192.168.2.100"`**

When I connect the ESP8266 to USB , the Arduino IDE will offer me two doors; one above and one more in **`"/usb/tty0"`**.

I imagine that without the ESP8266 connection by USB, Putty could connect through **`"/usb/tty0"`**.
However Putty can not access that port.

1. Should I use another serial monitor in place of Putty ? Any recommendations in particular - I use Ubuntu 15.10

2. You said:

**To know the serial port, I would:

- disconnect module from the PC,
- check the list of USB devices visible by O/S
- connect module back
- check what new port has been detected by the OS.**

What procedure should I do to find out which serial ports are available?

I usually use **`dmesg`** on my Linux; there is another way to check what is the available ports ?

odilonafonso, 2016-09-05:
Hello friends ... I could with Putty.
Thanks a lot for the help !!

closed 2016-09-05 by odilonafonso

odilonafonso, 2016-09-05:
I would love now to update the sketch in my ESP8266 for an internet connection. Not for the WiFi connection to my router.
It's possible ? Do you know any sample where I can guide me?
Thank you again !!

krzychb, 2016-09-05:
@odilonafonso,

Congratulations on making Putty work as desired :+1: 

To do OTA update over an internet connection you can try [HTTP Server](https://github.com/esp8266/Arduino/tree/master/doc/ota_updates#http-server) method. Sample sketch is  [httpUpdate.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/examples/httpUpdate/httpUpdate.ino).

For this one to work you need a web server that is able to run PHP scripts. 

Please also note that with [HTTP Server](https://github.com/esp8266/Arduino/tree/master/doc/ota_updates#http-server) method ESP module is _pulling_ update from the server. This is opposite direction than in [Arduino IDE](https://github.com/esp8266/Arduino/tree/master/doc/ota_updates#arduino-ide) or [Web Browser](https://github.com/esp8266/Arduino/tree/master/doc/ota_updates#web-browser) methods, where you are _pushing_ update to the module.

odilonafonso, 2016-09-06:
Hello @krzychb,

I would not have so easily without the help of you.

The code in the sample that u showed me - httpUpdate.ino - must be in the sketch to be updated via OTA?

And the file as the statement:

`ESPhttpUpdate.update ("http: //server/file.bin")`

It is the result of the compilation to be updated?

I do not know if I understand very well. Where to find documentation of these libraries:

`ESP8266WiFiMulti` and `ESP8266httpUpdate`

Thank again!

krzychb, 2016-09-07:
Hello @odilonafonso,

I am always glad to help :smile:

Regarding your questions:
* The `file.bin` should contain a new compiled sketch you intend to load to ESP using OTA. Instruction how to obtain `file.bin` is described in [Web Browser OTA method](https://github.com/esp8266/Arduino/tree/master/doc/ota_updates#web-browser).
* `httpUpdate.ino` – this is an example sketch demonstrating [HTTP Server OTA method](https://github.com/esp8266/Arduino/tree/master/doc/ota_updates#http-server).  Before loading it to ESP enter "SSID"” and "PASSWORD” to your Wi-Fi network  
* Restart ESP after loading this sketch. ESP will access `http://server` (you need to enter URL of your server here) and request  `file.bin`.
* After downloading `file.bin`, ESP will restart and run this new sketch.

I am not aware of other documentation for `ESP8266httpUpdate` class than contained in description of  [HTTP Server OTA](https://github.com/esp8266/Arduino/tree/master/doc/ota_updates#http-server) method.

`ESP8266WiFiMulti` is just another method to connect ESP to your Wi-Fi – check [this example](https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/station-examples.md). You can safely replace it with other code you typically use to connect ESP to Wi-Fi.

Hope this clarifies your concerns,

Krzysztof

odilonafonso, 2016-09-13:
Hello @krzychb,
Thank you again for help !!
These days that have passed did other things, today return to work in updating the sketch by internet.
Thank you for the tips!
I'll start by running the OTA update by Webserver on my Wifi network and then try to follow your instructions.
I will communicate with this new open issue: #2513

Vamshi253, 2016-09-27:
Hello @krzychb,

Thank you for sharing your knowledge.

Concerning OTA, I need to develop the protocol between the web server and the ESP8266.

In this case, ESP8266 should periodically check for updates on the server and if any new version is available, it should automatically updates itself.

I can be able to implement the program on the client side (ESP8266) using the class ESP8266httpUpdate in Arduino IDE.

I cannot be able to implement the setup of web server and configure the things on the server side.

Could you please elaborate the steps in detailed that I need to execute for server side implementation?

krzychb, 2016-09-28:
> Could you please elaborate the steps in detailed that I need to execute for server side implementation?

Hey @Vamshi253,

I am running web server on [Synology DS-414](https://global.download.synology.com/download/Document/DataSheet/DiskStation/14-year/DS414/Synology_DS414_Data_Sheet_enu.pdf) disk station. In my case there are the following steps to execute for server side implementation.

1. [Enable Web Station](https://www.synology.com/en-global/knowledgebase/DSM/tutorial/Application/How_to_host_a_website_on_Synology_NAS). The station has PHP alreay included as shown below:

  [image]

2. Save the PHP script provided in documentation of [Advanced Updater](https://github.com/esp8266/Arduino/tree/master/doc/ota_updates#advanced-updater-1) to the disk station in the folder used by Web Station for serving web pages.

3. In the PHP script update the table shown below with the MAC address and the name of file that contains binary code for your ESP. You may provide several lines of `"MAC" => "FileName`" to provide OTA updates to several modules.

  ```php
  $db = array(
      "18:FE:AA:AA:AA:AA" => "DOOR-7-g14f53a19",
      "18:FE:AA:AA:AA:BB" => "TEMP-1.0.0"
  );
```
4. Save compiled binary file in the `bin` sub folder of the Web Station's folder. The file should have `.bin` extension.

Vamshi253, 2016-09-28:
Hey @krzychb,

Thank you very much.

This would be very helpful for me.

Moreover, apart from implementation using disk station, do you know how to implement server side on pc using Apache server software with php support?

odilonafonso, 2016-09-28:
Wow @krzychb  !! This is a real disk station !! 

Hi @Vamshi253,

This week I am very busy next week if I have time I implement what you are asking for more "step-by-step."

But what our friend @krzychb explained above is all you need to implement, nothing more. In your apache with php. No mysteries.

Make a few attempts, if you can not, return with more specific questions.

The solution I'm finishing implement in my IoT's devices is A better suited for me. Through MQTT.

My device embedded IoT does not bother to check for a new release available; who cares that's me. If I a device meets my expectations working with a 2015 release, why will change the firmware?

When I want to change the firmware, I put it in a specific area of ​​my server (or my dreamhost server) and notice the device to "auto-update". 

No unnecessary queries. No php scripts. Of course, it is not as automated solution as the solution that you intend to implement, but is great for me! And consumes much less resources.

In time, why do not you open a specific topic for that matter? It will be easier arise more contributions related to what you want to do.
Do this and let us know if you wish.

Vamshi253, 2016-10-05:
Hi @krzychb and @odilonafonso,

Thank you very much.

Do anybody of you have a sample code based on advanced updater method (where if there is a version mismatch, then the firmware is updated) ?

krzychb, 2016-10-06:
@Vamshi253,

The code is among examples and you can execute advanced updater method after modifying a single line in it. Expanding on my [comment](https://github.com/esp8266/Arduino/issues/2482#issuecomment-250095983) above and carrying over with point 4:

  * Save compiled binary file in the `bin` sub folder of the Web Station's folder. The file should have `.bin` extension, for instance:

  ```  
    TEMP-1.0.0.bin
  ```
  * Open [`httpUpdate.ino`](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266httpUpdate/examples/httpUpdate) example sketch and replace:

  ```cpp
   t_httpUpdate_return ret = ESPhttpUpdate.update("http://server/file.bin");
  ```
  with

  ```cpp
  t_httpUpdate_return  ret = ESPhttpUpdate.update("http://server/update.php", "TEMP-0.9.5.bin");
  ```
  Note: This modification applies to module with MAC address `18:FE:AA:AA:AA:BB`.

  * Load module with updated sketch using a serial port. Restart the module by pressing reset or cycling power. It should then download `TEMP-1.0.0.bin` firmware and self restart.  

  Important: This new firmware should contain line like:

  ```cpp
  t_httpUpdate_return  ret = ESPhttpUpdate.update("http://server/update.php", "TEMP-1.0.0.bin");
  ```
  If not, it will keep restarting and downloading `TEMP-1.0.0.bin` over and over again.

Vamshi253, 2016-10-06:
Hi @krzychb,

Thank you very much.

Actually, based on my application, the ESP8266 is in a box and the box has to installed in a remote location where physical access to ESP8266 is not possible.

Whenever we change the firmware file on the server, ESP8266 should detect this change and updates itself.

Likewise, ESP8266 should periodically checks for updates on the server by itself and if any are available then it should update itself.

Is it a possible application by using advanced updater? If it is, can you please guide me on this?

Vamshi253, 2016-10-07:
Hi @krzychb and @odilonafonso,

I executed Simple Updater on my esp successfully on yesterday.

But when I try to execute simple updater again today, it's not working and indicating the following on serial output:

Exception (29):
epc1=0x402027bc epc2=0x00000000 epc3=0x00000000 excvaddr=0x00002000 depc=0x00000000

ctx: cont 
sp: 3ffef6b0 end: 3ffef900 offset: 01a0

>>>stack>>>
3ffef850:  fe000a2e feefeffe feefeffe 402027a3  
3ffef860:  3ffef890 3ffe8448 00000009 401004d8  
3ffef870:  feefeffe feefeffe 3fffb06c 3ffee8cc  
3ffef880:  3ffe843c 3ffee6f0 3ffe8448 40202871  
3ffef890:  3fffb08c 3fffb09c 3ffef870 3ffef8a0  
3ffef8a0:  3fffdad0 00000000 3ffee6f0 4020288c  
3ffef8b0:  3fffdad0 00000000 3ffee6f0 402021d7  
3ffef8c0:  402010ae 000003e8 000003e8 402010a3  
3ffef8d0:  00000000 00000000 00000001 40206351  
3ffef8e0:  3fffdad0 00000000 3ffee8c4 4020637c  
3ffef8f0:  feefeffe feefeffe 3ffee8e0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
ø

I tried a lot searching for this on internet. But I cannot be able to find the solution.

Could anybody of you help me regarding this?

odilonafonso, 2016-10-07:
Many apologies @Vamshi253, as I said before, I'm super busy these days, doing other things.
The procedure adopted to update my ESP is good for me.
As soon as I have time I will try to do what you intend to do.
Excuse me again, now I have no time.
Why do not you publish your attempts, show us your sketches? Perhaps then we can help you better.
It is hard to help you without knowing at what point are his attempts, not knowing what you have already achieved.
Good luck.

Vamshi253, 2016-10-11:
Hi @odilonafonso and @krzychb 

It's Okay. Thank you very much.

Concerning the client side (ESP8266), I have developed the following code:

#include <Arduino.h>
 
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
 
#define USE_SERIAL Serial
 
// ****************************************
// DON'T FORGET TO UPDATE EVERY VERSION!!!!
#define currentVersion 123          
// ****************************************
 
ESP8266WiFiMulti WiFiMulti;
 
void setup() {
    USE_SERIAL.begin(115200);
    for(uint8_t t = 4; t > 0; t--) {
        delay(1000);
    }
    WiFiMulti.addAP("NEXTRA8054", "02072897");
}
 
void loop() {
    // This loop will never exit - it is the main loop of the program
   // for (;;) {
        // Code here to read from sensors
        
        // Code here to send data to the IoT-cloud
        
        // Now check for updates
        CheckForUpdate(currentVersion);
        // Sleep for a minute, then loop back and take more readings
        delay(600000);
    //}
}
 
 
 
void CheckForUpdate(int){
    if((WiFiMulti.run() == WL_CONNECTED)) {
        HTTPClient http;
 
        http1.begin("http://192.168.1.134/fw/Version.txt");
        int httpCode = http1.GET();
        if(httpCode == HTTP_CODE_OK) {
            String payload = http.getString();
            int n = payload.toInt();
            if(n>currentVersion) {
            t_httpUpdate_return ret = ESPhttpUpdate.update("http://192.168.1.134/TEMP-1.0.0.bin");
            switch(ret) {
              case HTTP_UPDATE_FAILED:
                USE_SERIAL.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
                break;
             } //switch
          } // if version
        } // if code_ok    
    } // if wifi-connected
}

Generally, the above code just checks the version string on the server and if it is greater than existing version on the device, then OTA is executed.

But I want to incorporate the logic of OTA based on MAC Address, means every time if communication establishes between ESP8266 device and server, the ESP8266 device sends its MAC address to the server. Based on the mac address, the server provides the respective firmware file for the ESP8266 (OTA execution).

I know the function (wifi.macAddress) which outputs the MAC address. So I can use that on client side and send the mac address to the server.

But I don't know the logic and steps that I need to incorporate on the client side and server side for this type of OTA implementation.

Could anybody of you please help me regarding this?