esp8266/Arduino #2483: Memory leak on http.GET()

kireol, 2016-09-04:
#### Hardware
Hardware:		ESP-12e on Witty cloud and NodeMCU boards.
Core Version:      	Not sure where I find this

### Description
Memory leak when http.GET() is called

###Problem description
If I run the code with the .GET() commented out, I don't lose any memory.  If I uncomment the http.GET() line, I lose bytes every time according to ESP.getFreeHeap().  

### Settings in IDE

Library:                 Installed via "Additional Boards manager"  http://arduino.esp8266.com/stable/package_esp8266com_index.json

Module:  			NodeMCU 0.9 (ESP-12 Module)
Flash Size: 		4MB (3MB Spiffs)
CPU Frequency:	80Mhz
Flash Mode:		I dont see this anywhere
Flash Frequency:	I dont see this anywhere
Upload Using:		SERIAL
Reset Method:	Not sure

### Sketch

```
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
ESP8266WiFiMulti WiFiMulti;
void setup() {
  Serial.begin(115200);
  WiFiMulti.addAP("myssid", "mywifipass");
}

void loop() {
     long mem = ESP.getFreeHeap();
  Serial.print("freeMemory()=");
    Serial.println(mem);

  if((WiFiMulti.run() == WL_CONNECTED)) {
    HTTPClient http;
    http.begin("http://www.myprivatejson.url");
    int httpCode = http.GET();
    http.end();
  }
}
```

### Debug Messages

> freeMemory()=40472
> freeMemory()=40288
> freeMemory()=40104
> freeMemory()=39920
> freeMemory()=39736
> freeMemory()=39552
> freeMemory()=39368
> freeMemory()=39392
> freeMemory()=39184
> freeMemory()=39208
> freeMemory()=38880
> freeMemory()=38816
> freeMemory()=38632
> freeMemory()=38448
> freeMemory()=38472
> freeMemory()=38264
> freeMemory()=38288
> freeMemory()=38080
> freeMemory()=37896
> freeMemory()=37712
> freeMemory()=37528
> freeMemory()=37552
> freeMemory()=37344
> freeMemory()=37160
> freeMemory()=36976
> freeMemory()=36792
> freeMemory()=36608
> freeMemory()=36424
> freeMemory()=36240
> freeMemory()=36056
> freeMemory()=35872
> freeMemory()=35688
> freeMemory()=35504
> freeMemory()=35320
> freeMemory()=35344
> freeMemory()=35136
> freeMemory()=34952
> freeMemory()=34768
> freeMemory()=34584
> freeMemory()=34400
> 

messages here
```

krzychb, 2016-09-04:
Hi @kireol 

I run this sketch replacing  `http://www.myprivatejson.url` with `http://www.esp8266.com/` and did not observe any memory leakage.

```
freeMemory()=46624
freeMemory()=46864
freeMemory()=46624
freeMemory()=46864
freeMemory()=46864
freeMemory()=46744
freeMemory()=46624
freeMemory()=46864
freeMemory()=46864
freeMemory()=46864
freeMemory()=46864
freeMemory()=46864
freeMemory()=46744
freeMemory()=46624
```

I did this test with using latest satble esp8266 / Arduino core 2.3.0 and NodeMCU 1.0 (ESP-12 Module).  

For Arduino IDE, you can check the core version using in Boards Manager as below.

[image]

kireol, 2016-09-04:
Thanks. It sounds like the good news is it is my setup then, and not a bug in master.  Somehow I'm running 1.0.0? 

Is this my core?
[image]

I followed the readme.  Here's how I installed it.
[image]

Here's my libraries dir.
[image]

It sure looks like it is me.  But what did I screw up?

kireol, 2016-09-04:
ahhhh.  Found it.

Boards manager was not detecting a new version.

I updated to 2.3.0 and the bug is gone.

Sorry for the trouble.

closed 2016-09-04 by kireol

jeancode, 2018-01-13:
Hola que tal amigos  yo tambien me he topado con este problema con la ram lo he solucionado con una manera simple cada vez que se hace la petición por defecto  HTTPClient abre una instancia nueva ademas de la que ya existe

SOLUCION:

HTTPClient http;

void  setup(){
	
      //memory fail reques
	http.setReuse(false);

	//memory full
	http.setReuse(true);

}