esp8266/Arduino #2488: Watchdog timeout during ESPhttpUpdate

tomkcook, 2016-09-04:
Updating from an HTTP server always fails for me with a watchdog timeout, apparently while downloading the new firmware image.

### Basic Infos

#### Hardware
Hardware:			ESP-12F
Core Version:      	2.3.0

### Description

ESPhttpUpdate.update trips the watchdog timer.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		160Mhz
Flash Mode:			dio
Flash Frequency:	80Mhz
Upload Using:		Serial (the current image is uploaded by serial, I'm trying to get OTA updates working)
Reset Method:		ck

### Sketch

Here's the relevant bit:

```cpp
UPDATE_HOST="my.update.host.com";
CURRENT_VERSION="1.0.3";

			Serial.println("Beginning OTA update from " UPDATE_HOST);
			auto result = ESPhttpUpdate.update(UPDATE_HOST, 80, "/update", CURRENT_VERSION);
			if( result == HTTP_UPDATE_FAILED ) {
				Serial.println("Update failed");
			} else if( result == HTTP_UPDATE_NO_UPDATES ) {
				Serial.println("No update required");
			} else if( result == HTTP_UPDATE_OK ) {
				Serial.println("Update succeeded");
			} else {
				Serial.print("Unknown HTTP update response: ");
				Serial.println(int(result));
			}

```

### Debug Messages

```
Beginning OTA update from my.update.host.com
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 401184
[httpUpdate]  - MD5: 73a60a06914be211115f3a7257ddbe29
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 2744320
[httpUpdate]  - current Sketch Size: 401184
[httpUpdate]  - current version: 1.0.3
[httpUpdate] runUpdate flash...

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

As you can see, the server is returning a valid set of headers (at least) and the updater apparently starts reading the binary from the server, but the watchdog times out while this is happening.

I thought that the `yield()` at line 341 of Updater.cpp should prevent this from happening; is the problem just that my HTTP server is so slow it never gets that far?

pieman64, 2016-09-04:
You haven't provided a sketch so we don't know what line 341 or the other few hundred lines contain.

WereCatf, 2016-09-04:
@tomkcook What's this Updater.cpp you are talking about? I know of https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp but there isn't a single call to yield() there.

tomkcook, 2016-09-05:
Line 366 of that file (ESP8266httpUpdate.cpp) calls `Update.writeStream(in)`.  `Update.writeStream` is on line 312 of `Updater.cpp`, and line 341 of Updater.cpp has `yield();`.

The sketch where I see this is rather involved; I'll reduce it to a test case today.

tomkcook, 2016-09-05:
In other words, _this_ Updater.cpp: https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Updater.cpp

devyte, 2017-10-13:
@tomkcook Is this issue still relevant with latest git?

tomkcook, 2017-10-19:
I don't know, I'm sorry.  I'm not working on this any more.

devyte, 2018-11-12:
Closing due to age and lack of feedback.

closed 2018-11-12 by devyte