esp8266/Arduino #2490: Serial output stops after a few lines on ESP8266-01

RogerSchaefer, 2016-09-06:
I have a rather large sketch that reads sensors using I2C and displays results on an OLED	display and also sends the results out on the Serial interface.  After a few lines the Serial stops but the program continues running displaying data results on the OLED.

The exact same sketch runs perfectly on a NodeMCU outputting everything programmed to the Serial port.

The problem seems to start with the initialization of the OLED using the Adafruit_SSD1306.h library.

It looks like the NodeMCU v1.0 and the Generic ESP8266-01 are compiling differently.

Roger Schaefer

devyte, 2017-10-13:
@RogerSchaefer is this issue still valid with latest git?
You said: 
> It looks like the NodeMCU v1.0 and the Generic ESP8266-01 are compiling differently.

Did you compare the binaries to arrive at that conclusion?

devyte, 2017-12-29:
Closing due to no feedback in over 2 months, and age.

closed 2017-12-29 by devyte