esp8266/Arduino #2491: Tone causes crash with frequency = 0

cranphin, 2016-09-07:
It's not uncommon to use tone with frequency = 0 to indicate 'silence/no note'.
Example: https://github.com/arduino/Arduino/blob/master/build/shared/examples/02.Digital/toneMelody/toneMelody.ino
This crashes right now, I expect because of a division in zero in: https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Tone.cpp
>  timer1_write((clockCyclesPerMicrosecond() * 500000) / frequency);
Admittedly the behavior on Arduino for frequency = 0 doesn't seem to be documented, and I expect on some other cores it would also crash or do weird things..

Razpudding, 2016-12-08:
I second this issue. I worked around it by checking if frequency is '0' and if so setting it to '1'. If someone has an actual solution I'd love to hear it.

pfeerick, 2016-12-20:
Over on the Oak we've also just experienced this issue as well, as it is essentially based on the ESP8266 core.  In the issue [I opened over there](https://github.com/digistump/OakCore/issues/75) to document that it's a known problem, I proposed the following solution to work around the issue, but only at the user sketch level... it doesn't really fix the problem as it results from the implmentation, but it at least prevents the sketch from crashing...

Basically, Instead of

```tone(3, melody[thisNote], noteDuration);```

to use

```if (melody[thisNote] == 0)
{
  noTone(3);
  delay(noteDuration);
}
else
{
  tone(3, melody[thisNote], noteDuration);
}```

devyte, 2017-10-13:
Code is already merged, closing.

closed 2017-10-13 by devyte