esp8266/Arduino #2493: More Details needed on: .text' will not fit in region `iram1_0_seg'

stefanbode, 2016-09-07:
### Basic Infos

#### Hardware
Hardware:			NodeMCU 1.0

### Description

Compiler error after adding:
analogwrite() inside mqtt_callback function

### Settings in IDE Board version 2.3.0

Module:  			Nodemcu 1.0ß
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

I do understand already after reading different topics about this issue that the 32k RAM area is full. I also have found the ICACHE_RAM_ATTR and ICACHE_FLASH_ATTR to tag functions.

Anyhow both do not make a change on my mqtt callback. Does not compile anymore. How can I debug what function is already in the iRAM and do I have to option to move these to flash. As far as I can see the functions in my sketch go all to flash because without the analogwrite I can add the ICACHE_FLASH_ATTR to functions without a change. Also removing the ICACHE_RAM_ATTR from the mqtt_callback does not free up space. The board version 2.2.0 does compile but have other errors in the mqtt stack that are fixed in 2.3.0 therefore I want to stick with this

Question:
How to see actual occupation of IRAM?
How to predict which function go into IRAM? (does Serial.print in mqtt callback have any impact; does not look like)

devyte, 2017-10-13:
@stefanbode is this issue still valid with latest git?

stefanbode, 2017-10-13:
I discontinued the project and do now contribute to TASMOTO. Therefore the issue is neither relevant, nor can I check if with the current version my build now works.

devyte, 2017-10-13:
Closing as no longer relevant.

closed 2017-10-13 by devyte