esp8266/Arduino #2494: How to lock the firmware hex file ?

waqas-ogc, 2016-09-07:
#### Hardware
Hardware:			WROOM-2 based on ESP8266
 
### Description
I'm designing a commercial product and wants to protect my hex file from reading back. Is there a way for this ? ,like in other microcontrollers there are fuses to lock the firmware.

igrr, 2016-09-07:
No, there is no way to do that — flash is external and can be read back trivially.
You may want to look at the ESP32 instead. It has flash encryption hardware which can help keep code in external flash secure.

closed 2016-09-07 by igrr

human890209, 2018-08-28:
Hi, I'm wondering if using ESP8285 which has a built-in 1M flash, is it possible to protect the hex file?
Cause ESP8285 is almost the same as ESP8266, I put my question here.
I'm now using ESP8266 and got half of my work done, then the code protection problem comes.
I'm not using a lot of Flash right now, I think 1M is enough for my project.
Hope someone could see this. @igrr

igrr, 2018-08-28:
In ESP8285 the contents of internal flash can be read out via UART interface using esptool.py, or by connecting a programmer to SPI flash directly (it is connected to the outer package pads). So it is no different from ESP8266 in this aspect.

human890209, 2018-08-29:
Hi, @igrr. Thanks for your answer. 
ESP8285 seems no improvement at all...
Is there any workaround by software? Use ChipID as a key or something?

Suraj151, 2019-05-07:
Is there any workaround on this. i am not expecting that code in esp8266 should be protected but at least how can i avoid esp8266 chip from flashing once again after production bin is flashed.

rcarlson01, 2019-10-03:
The best solution is to use an encryption tools for the ESP8266 with custom bootloader from https://www.keeloc.com.  It costs $1.00 to register and encrypt each device you want to code protect.  Your unecrypted binary of firmware isn't shared with even the company doing the encryption.  Basically you download their tools and it will send the proper unique ids and security bits and then these bits are uploaded to their server which returns the encryption key.  Encryption is performed local on your linux or windows desktop with the AES that is returned.  All this is done in a chain operation of either a batch file for windows or a bash script for linux.  Everything is done quickly in the chain with about a 20 second time frame for all operation, including that the encrypted code and bootloader are automatically loaded onto the targeted ESP8266.  To truly protect your code you are recommended to seal off the serial port pin(s) by fusing it out, that is putting some code in your firmware after it first loads up that will drive the serial port pins low impedance and high voltage.  This pin is then provided a path to ground.  By doing this you prevent the serial port usage which can be a disadvantage, but also this prevents hackers from getting information about the unique ID's and secuirty bits to try to decrypt your firmware.  It's not totally necessary to blow the pin, but then you allow a vector to possibly derive your AES key and decrypt your firmware by doing that.  Truth is it would be better to run it with a bit of protection than none even if you decide not to fuse out the serial pins.