esp8266/Arduino #2495: Software SPI for ESP8266 - Pointer Please

SrikanthBala, 2016-09-07:
Hardware:		ESP-12E
Core Version:      	2.3.0

Hi guys! Once more I'm stuck and need a pointer in the right direction. I'm using a ESP12E on NodeMCUV1.0 dev board. The ESP functions as a SD webserver and serves as a front end for my application. The ESP is also connected via serial port to a 3.3V Arduino  mini which communicates to a DS18B20 temp sensor, a 8x8 matrix display, a DS3231( for those many times the ESP cannot connect to a NTP server to get the time), and acts as a datalogger by sending the temp data to the ESP via serial along with a time stamp and the ESP logs this onto the SD card as a csv file. At the turn of the month, it opens a new monthly file and logs there. (It also simultaneously updates a link to this csv file onto another htm file). This HTML file is a download link to these monthly csv files which I can download from my computer to view. This arrangement works fine. ...........My problem starts now.........

I need to have my ESP program my Arduino using SPI (using a ArduinoISP like sketch)  or implement a Serial protocol like STK500 on my own so that the bootloader can be used for programming via Serial. I tried looking for the latter but nothing concrete turned up. **On the other hand the ArduinoISP is a proven working example.** My Arduino and ESP are connected to reset each other when needed (via transistors to prevent accidental reset when Arduino reboots). 

But my SPI pins are taken up by the SD card. True I could connect the SPI pins to the Arduino too and pulling the RST pin low will put it in SPI mode( I could keep the SS of SD card inactive). However the flash file would be 30KB and I couldn't store this in RAM I suppose. So I'll have to switch between reading the .bin (hex) file from the SD card (where I would presumably store it after downloading it from the net) . This introduces a problem. When I switch SPI select pins, if I hold the Arduino RST low I would write junk onto the Arduino as SCK, MISO and MOSI would be pulsing while reading the card. (I believe the RST is some sort of Chip Select or SS pin for the SPI while programming). I don't think programming a page and then releasing the RST pin while I read the SD file and then restarting the programming process is an option.

So I need to use other ESP GPIO pins for a Software SPI. I have the GPIO 0,2 and 16  pins (with their known problems during the startup process) available. My GPIO16 is not tied to RST of the ESP as it is in some older boards. 

My SD card is connected as follows (CS to GPIO4, SCK to GPIO 14, MOSI to GPIO13 and MISO to GPIO 12) as per https://www.reddit.com/r/esp8266/comments/48ib9g/connecting_a_sd_card_to_my_esp8266/
**and not as** the for example mentioned in the SDesp8266  ReadWrite Example (i.e.  MOSI - pin 11, MISO - pin 12,  CLK - pin 13, CS - pin 4), and my connection is working

So finally my question is how can I use a software SPI using pins GPIO 0, GPIO 2 and GPIO 16 to program my Arduino or to read the SD card. I already have another pin (GPIO 15 tied to a NPN transistor to reset my Arduino as and when desired. GPIO 15going low at boot up will not affect my NPN transistor and so will not accidentally rest my Arduino, not that it matters if it does in my particular setup) 
Please point me in the right direction. Thanks for reading thru the long description.

Regards
SBala

### Settings in IDE

Module:  			ESPNodeMCU V1.0
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		OTA + SERIAL
Reset Method:		ck / nodemcu

SrikanthBala, 2016-09-07:
I tried compiling my sketch using a byte array of size 32768. It compiled fine leaving some 9000 bytes for local variables but gave a warning "Low memory available, stability problems may occur". This would actually solve my problem as I can read the entire SD content in one go and write it using the Arduino ISP sketch. 
What stability problems should I expect ? Also couldn't I release the memory dynamically after my programming is done to avoid these problems ?

WereCatf, 2016-09-07:
The "difference" between your wiring of the SD-card and the SDesp8266ReadWrite-example is because...you're referring to the pins by GPIO-number, the example refers to physical pin-numbers of the ESP8266-module -- that is to say, you are connecting to the exact same pins, except for CS.

Also, yes, you could dynamically allocate and release the 32KB buffer using e.g. malloc() and free(), but then you wouldn't have much left if in the future you wanted to add some functionality to it. If it works for you and you feel that's fine, then by all means, go for it. Another option could be to e.g. read the 32KB-file to an I2C FRAM-package (see e.g. https://learn.adafruit.com/adafruit-i2c-fram-breakout/overview ) and during flashing of your Arduino read from there -- I2C requires only 2 GPIO-pins, so you should be good to go in that regards. Consider your options and proceed as you like.

WereCatf, 2016-09-07:
Actually, I'll have to correct myself: it looks like the pin-numbers in the example are not even for ESP8266 in the first place. They are for Arduino Uno.

SrikanthBala, 2016-09-07:
Thank you for the prompt help. I'll definitely look up the I2C FRAM option. 

Meanwhile, correct me if I am confused but I believe I have got it all wrong. I cannot flash my AVR pro mini from a hex file simply by using the code from ArduinoISP. I still need a STK500 protocol implementation. Looking at the ArduinoISP code shows that it is simply transferring bytes from avrdude got via serial to the SPI bus. I need to actually implement the STK500 protocol myself and do partly what avr dude does, to flash directly from a hex file. I think I will need to use something like described in https://github.com/SavchukSergey/FlasherCore  and then I could simply flash using a serial port (which I already have connected) using the bootloader on the Arduino. 
I think I should research  a bit more  into how to program an AVR directly.
I'm sorry I posted too quickly without adequate research

Regarding the pin connections being for the Arduino, I think I got them from the ReadWrite sketch of the ESP8266 but I guess I'm wrong. Thanks for correcting me.

Thank you all once again for the wonderful ESP8266 Arduino IDE implementation and your prompt help for newcomers like me.

WereCatf, 2016-09-07:
You can just adapt the ArduinoISP-code, but replace all the parts that would read the data from serial with code that reads the data from the I2C FRAM, and strip out everything else related to serial. STK500 and the likes are for communication between the programmer and the computer, but you already have the hex-file on the ESP, so you don't need that part at all! STK500 has nothing to do with the actual flashing of the target-device, that part is still either ISP- or parallel-programming.

SrikanthBala, 2016-09-07:
As happens often in the world, I have been re-inventing the wheel. 
Mr Kevin Osborn and Mr Justin Shaw  have already achieved what I was looking for 5 yrs back. They have implemented avrdude on an Arduino so that it can program another from a SD card. They call it BootDrive. http://baldwisdom.com/bootdrive/
I just have to port it to the ESP and I have what I need. No more searching for another SPI port.

Did not find it earlier as I did not look long enough in the arduino forum for the right thing.

Will close this issue now. After I succeed, I'll post my experience here if anybody else needs it.

Thanks again 

SBala

SrikanthBala, 2016-09-07:
Sorry WereCatf. Didn't see your post before I posted and closed. You mean to say that the hex file can just be written directly and to the Arduino ? I though the hex file was the actual binary file and the STK500 protocol defined how to send the appropriate commands to send the hex file page by page to the arduino memory. So I thought I would need to implement that

closed 2016-09-10 by SrikanthBala