esp8266/Arduino #2497: Losing contact with DHT11 after 1 reading

cjwolff, 2016-09-08:
### Hardware
Hardware:			NodeMCU 1.0 (ESP-12E Module)

### Description
Hello,
I'm having an issue that kept me up late that I can't seem to resolve reading the docs and following the tutorials correctly.  I'm using the latest version of the adafruit DHT library, a DHT11 sensor, and a JSN-SR04T.  The DHT11 gives one humidity and temperature reading each time the nodemcu is reset or uploaded then quits as shows in the serial console debug messages below.  The JSN-SR04T maintains distance readings even though the DHT11 stops.

I've tried different delays as I've heard that the DHT11 needs to rest between readings without resolution.  

Any advice would be appreciated.  Thank you.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Frequency:	?40Mhz?

### Sketch

#include DHT.h
#include NewPing.h
#include ESP8266WiFi.h

const char* MY_SSID = "****"; 
const char* MY_PWD = "****";

#define TRIGGER_PIN  5  // Arduino pin tied to trigger pin on the ultrasonic sensor.
#define ECHO_PIN     4  // Arduino pin tied to echo pin on the ultrasonic sensor.
#define MAX_DISTANCE 200 // Maximum distance we want to ping for (in centimeters). Maximum sensor distance is rated at 400-500cm.
#define DHTPIN 14
#define DHTTYPE DHT11

NewPing sonar(TRIGGER_PIN, ECHO_PIN, MAX_DISTANCE); // NewPing setup of pins and maximum distance.
DHT dht(DHTPIN, DHTTYPE, 15);

void setup() {
  Serial.begin(9600); // Open serial monitor at 115200 baud to see ping results.
  dht.begin();
  connectWifi();
}

void loop() {
  delay(5000);
  float h = dht.readHumidity();
  float t = dht.readTemperature();
  float f = dht.readTemperature(true);
//    if (isnan(h) || isnan(t) || isnan(f)) {
//    Serial.println("Failed to read from DHT sensor!");
//    return;
//}
    // Compute heat index in Fahrenheit (the default)
  float hif = dht.computeHeatIndex(f, h);
  // Compute heat index in Celsius (isFahreheit = false)
  float hic = dht.computeHeatIndex(t, h, false);

  Serial.print("Humidity: ");
  Serial.print(h);
  Serial.print(" %\t");
  Serial.print("Temperature: ");
  Serial.print(t);
  Serial.print(" *C ");
  Serial.print(f);
  Serial.print(" *F\t");
  Serial.print("Heat index: ");
  Serial.print(hic);
  Serial.print(" *C ");
  Serial.print(hif);
  Serial.println(" *F");
  delay(5000);
  Serial.print("Object: ");
  Serial.print(sonar.ping_cm()); // Send ping, get distance in cm and print result (0 = outside set distance range)
  Serial.println("cm");
  
}

void connectWifi()
{
  Serial.print("Connecting to "+*MY_SSID);
  WiFi.begin(MY_SSID, MY_PWD);
    while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
    }
  
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println("Netmask: ");
  Serial.println(WiFi.subnetMask());
  Serial.println("Gateway: ");
  Serial.println(WiFi.gatewayIP());
  delay(1000);
}
>

### Debug Messages

WiFi connected
IP address: 
192.168.0.8
Netmask: 
255.255.255.0
Gateway: 
192.168.0.1
Humidity: 37.00 %	Temperature: 25.00 *C 77.00 *F	Heat index: 24.52 *C 76.14 *F
Object: 137cm
Humidity: nan %	Temperature: nan *C nan *F	Heat index: nan *C nan *F
Object: 137cm
Humidity: nan %	Temperature: nan *C nan *F	Heat index: nan *C nan *F
Object: 137cm
Humidity: nan %	Temperature: nan *C nan *F	Heat index: nan *C nan *F

consolacion, 2016-10-27:
Happening to me too, but I noticed that he readings go fine... until I added code to the program, then suddenly the readings would fail. Firts reading OK, just failure after that.

I do not know if it is a solution or a workaround, but here is a tip that improved it a lot, look for the reply of JayEmSydney  http://www.esp8266.com/viewtopic.php?f=32&t=4210&p=57237#p57237

devyte, 2017-10-13:
@cjwolff I am using DHT.h myself, on latest git, and I can't reproduce this. I should note that I do see failed readings every so often, but they are very few, as mentioned in the forum discussion above. Certainly not like in the original post described.
Closing as can't reproduce.

closed 2017-10-13 by devyte