esp8266/Arduino #2499: OTA (HTTP client mode) running out of memory and crashing when using TLS (https)

silbe, 2016-09-10:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	git master (commit 4897e000)

### Description

When modifying the ESP8266httpUpdate example "httpUpdate" to use an https URI, the ESP8266 crashes with a null pointer dereference (Exception 29 = StoreProhibitedCause,  excvaddr=0). Debugging suggests we're running out of DRAM, even though the sketch is not doing anything other than OTA. The server uses a 2048-bit RSA key with a certificate issued by letsencrypt.org.

The null pointer dereference is already tracked as #2201; this ticket is about running out of memory in the first place. The failure is actually very similar, but in #2201 the server used a 4096-bit key (which **could** be considered excessive), whereas we're using the absolute minimum currently permitted by most CAs and major browsers. If it doesn't even work with a 2048-bit key and an otherwise empty sketch, it's pretty much useless for practical purposes.

### Build system

Build system:		makeEspArduino (commit 18d85bc5)
Module:  			Generic ESP8266 Module (BOARD=generic)
Flash Size: 		4MB/1MB (FLASH_DEF=4M1M)
CPU Frequency:		unmodified
Flash Mode:			unmodified
Flash Frequency:	unmodified
Upload Using:		SERIAL
Reset Method:		manual (power switch)

### Sketch

```cpp

/**
 * httpUpdate.ino
 *
 *  Created on: 27.11.2015
 *
 */

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {

    // USE_SERIAL.begin(115200);
    USE_SERIAL.begin(57600);
    // USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

//    WiFiMulti.addAP("SSID", "PASSWORD");
    WiFiMulti.addAP("bbox", "PASSWORD");

}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {
[... 229 lines cut ...]
0x4020576c: SSLContext::_readAll() at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:587
0x4020576c: SSLContext::_readAll() at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:587
0x4020592d: SSLContext::available() at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:587
0x4020597b: WiFiClientSecure::connected() at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:587
0x4020760e: HTTPClient::connected() at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp:245
0x40204880: HTTPClient::handleHeaderResponse() at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp:197
0x40202918: String::~String() at /lhome/sascha.silbe/hardware/esp8266/Arduino/cores/esp8266/WString.cpp:519
0x40204648: HTTPClient::sendHeader(char const*) at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp:197
0x40204450: HTTPClient::connect() at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp:197
0x40204978: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp:197
0x40202b65: String::operator=(char const*) at /lhome/sascha.silbe/hardware/esp8266/Arduino/cores/esp8266/WString.cpp:519
0x40202434: operator new[](unsigned int) at /lhome/sascha.silbe/hardware/esp8266/Arduino/cores/esp8266/abi.cpp:57
0x402042cb: HTTPClient::collectHeaders(char const**, unsigned int) at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp:197
0x402049aa: HTTPClient::GET() at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp:197
0x4020392b: ESP8266HTTPUpdate::handleUpdate(HTTPClient&, String const&, bool) at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp:36
0x4010020c: _umm_free at /lhome/sascha.silbe/hardware/esp8266/Arduino/cores/esp8266/umm_malloc/umm_malloc.c:1287
0x4010068c: free at /lhome/sascha.silbe/hardware/esp8266/Arduino/cores/esp8266/umm_malloc/umm_malloc.c:1733
0x4022ebea: wifi_get_opmode at ??:?
0x40203c35: ESP8266HTTPUpdate::update(String const&, String const&, String const&) at /lhome/sascha.silbe/hardware/esp8266/Arduino/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp:36
0x40202a3e: String::String(char const*) at /lhome/sascha.silbe/hardware/esp8266/Arduino/cores/esp8266/WString.cpp:519
0x4020338d: loop at /home/sascha.silbe/hardware/esp8266/https-update-minimal/https-update-minimal.ino:47
0x40202605: HardwareSerial::setDebugOutput(bool) at /lhome/sascha.silbe/hardware/esp8266/Arduino/cores/esp8266/HardwareSerial.cpp:177
0x40202260: loop_wrapper at /lhome/sascha.silbe/hardware/esp8266/Arduino/cores/esp8266/core_esp8266_main.cpp:123
0x40100718: cont_norm at /lhome/sascha.silbe/hardware/esp8266/Arduino/cores/esp8266/cont.S:109
```

davisonja, 2016-12-13:
I had a slightly more involved sketch (fetches time via NTP and uses NeoPixel library to drive an array of WS LEDs with the number of days until Christmas.
It's been running happily for a few weeks with an https update URL (also 2048-bit letsencrypt :); however, I've also taken that sketch and added a bunch of other random stuff which subsequently ran into the same issue. I've seen suggestion that the SSL code is running fairly close to the limits of RAM, but if it's common to for it to crash with only mildly complex sketches then it is a bit unhelpful.
Tracing through is on my list... have you had any further insights?

silbe, 2016-12-21:
FWIW, my systems are running reasonably stable after switching to a custom implementation that fetches only a cryptographically secure checksum over HTTPS and performs the actual update over HTTP. This isn't optimal for privacy, but still provides protection against firmware modification.

davisonja, 2016-12-22:
This feels like it's starting to drift into #2103 which I'm quite keen to get sorted (it's the follow up plan after #905, which I've pretty much completed). With a signed update image and a key on the device you should be fine to use http regardless...

teji24, 2017-02-28:
I've got the same problem as you. But with a very small original sketch (the sketch making the upgrade), it work to HTTPS OTA update. As you said before, it's probably a memory issue. 
Is it possible to imagine a way to make a (dirty) clean of ram just before the update ? We don't need any more variables or others when the upgrade fire, ... and after the upgrade, the ESP will reboot.... It's maybye a way to make it run "before a best solution" ... ?

devyte, 2017-10-13:
@silbe is this issue still valid with latest git?

alexodus, 2017-10-16:
I'm using last rc (2.4.0-rc2) and this issue is still present.

igrr, 2017-10-17:
I think there should be enough RAM at this point for OTA-over-HTTPS to be possible, as axTLS memory usage was trimmed down by several kB since 2.3.0. However the issue likely is that there isn't a single contiguous block of 16kB at the time it is needed. I have a work-in-progress implementation of memory pool for axTLS, which should help with this, but it's not ready yet and i will try to finish it once 2.4.0 is released.

devyte, 2017-10-17:
@igrr A crazy thought that keeps nagging at me: instead of trying to hold the entire 16KB buffer in RAM, do you think it would be possible to mimic a 16KB buffer in a manner similar to a computer swap file? I mean have the data on flash or SPIFFS, and hold in RAM only e.g.: 4KB at a time. You could put an accessor in front to make it look like a buffer. When you to access beyond the 4KB in RAM, you swap the 4KB block with the corresponding block in the flash.
In other words, have a sort of swap-buffer.
Of course it would be _a lot_ slower, especially because it would be software based without an mmu, but if it's workable, it would relax the RAM requirements a ton...

devyte, 2017-10-17:
In my mind usage would look something like this:
```
SwapBuffer buffer(16*1024, 4*1024): //16KB total, 4KB in RAM
buffer[0] = 1;
buffer[1] = 1;
...
buffer[4095] = 1;
buffer[4096] = 1; //this tries to write beyond the block => swap the current block with a block in th flash => write to index 0
buffer[4097] = 1;  //this tries to write within the current block, so write to index 1
...

uint8_t tt = buffer[33]; //try to read from a block different than the current => swap the block with the one needed, then return index 33

```

igrr, 2017-10-17:
It's possible, i think. The question is how many flash sectors you allocate for swap space. Too few, and you kill your flash pretty fast. Too many, and your filesystem can no longer fit the files you need. Also this is going to be very slow. Assuming a 16kB TLS fragment buffer, that's 4 write-erase cycles to fill it with received encrypted data, and another 4 write-erase cycles to decrypt it.
8 write-erase cycles would take around 250ms, which would be added to roundtrip latency of your network connection, reducing download speed.

For those who find this issue a showstopper, deploying a Haproxy or nginx instance can be solution. It would be configured to support TLS fragment size negotiation extension. Then ESP8266 can tell that it wants to use 4kB TLS fragments, and 16kB buffer would not be needed at all.