esp8266/Arduino #2502: Serial.flush() have a problem

jyzhkj, 2016-09-11:
void loop()
{
digitalWrite(12, 1);
Serial.print("abcdffew");
Serial.flush();
digitalWrite(12, 0);
delay(100); // wait for a second
}

see the 12 pin when down, the last byte is not Send End.

so I try to change the HardwareSerial.cpp to:

void HardwareSerial::flush()
{
if(!_uart || !uart_tx_enabled(_uart)) {
return;
}

uart_wait_tx_empty(_uart);
long tt =uart_get_baudrate(_uart);
tt=12000000/tt;

delayMicroseconds(tt);
}

plinioseniore, 2016-10-30:
Hi @jyzhkj from your code looks like you are using an USART to RS485, I've found the same behavior and the last byte is not send as the flush doesn't empty the send queue before moving to the next instruction.

fishersolutions, 2017-02-13:
I have also found the same issue.
Is there a fix other than adding in your own delay?

d-a-v, 2018-10-12:
@jyzhkj This 2 years old issue seems to have been solved a year ago by #3714.
Is it OK, can you close it ?