esp8266/Arduino #2503: OTA No response from device.. [ask]

irwanayeah, 2016-09-11:
hi, a have problem

> Uploading .pioenvs\esp07\firmware.bin
> 08:36:10 [DEBUG]: Options: {'esp_ip': '192.168.137.36', 'host_port': 26909, 'image': '.pioenvs\\esp07\\firmware.bin', 'host_ip
> ': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
> 08:36:10 [INFO]: Starting on 0.0.0.0:26909
> 08:36:10 [INFO]: Upload size: 249088
> 
> 08:36:10 [INFO]: Sending invitation to: 192.168.137.36
> 08:36:10 [INFO]: Waiting for device...
> 08:36:20 [ERROR]: No response from device

sembsa, 2016-09-16:
I have this same problem :(

sembsa, 2016-10-10:
Hello,
Check your firewall settings for Python :)

EUA, 2016-12-06:
No firewall here, using Linux 64 bit.
Looks like recent update(s) kills the OTA

EUA, 2016-12-06:
Found issue:
When you select "Generic ESP8266 module" and "512KB FLASH size **with 64 SPIFFS**", which is default, at generated firmware OTA doesn't work. 512KB FLASH without SPIFFS work proper.

me-no-dev, 2016-12-06:
OTA on 512K is not supported. Not enough space

EUA, 2016-12-08:
But it is possible to do it. I use 512KB OTA without any issues at latest release. There are no problems.

You have better to put some compilation flag that gives error while compiling with 512KB**+64KB SPIFF** & OTA

Thijxx, 2017-05-03:
This bug is still there

devyte, 2017-10-13:
In recent releases, the binary size has increased beyond the point where OTA can be supported on a 512KB device. This is in part due to the Espressif SDK increasing in size, in part due to the Arduino code increasing in size.
On a 512KB device, the max sketch size for OTA support would be of the order of 230KB, less with SPIFFS. I just built a very simple sketch, and the binary size was over 300K. 
Also, I seem to remember a comment by Espressif that not even the official AT firmware would fit on 512KB devices anymore.
Closing.

closed 2017-10-13 by devyte

thepotoo, 2018-02-12:
Is it possible to change the error message here?  I have no idea if I'm doing something wrong trying to flash OTA upgrades to my ESP8266-01s, or if this is something that will never work.

At least for the issue I'm seeing, this is the part that's failing:
```
    sock.settimeout(10)
    connection, client_address = sock.accept()
    sock.settimeout(None)
    connection.settimeout(None)
```

I tweaked the code to print the error message, and it looks like it's timing out (no surprise there).  Increasing the timeout didn't help either.