esp8266/Arduino #2505: configTime and time seem to ignore DST

tzapu, 2016-09-12:
### Basic Infos
configTime and time seem to ignore DST

#### Hardware
Hardware:			ESP-12
Core Version:      	latest master

### Description
doing a config time, with DST offset does not seem to be actually doing anything to returned time. Time zone offset applies alone.

### Sketch

```cpp

#include <Arduino.h>

void setup() {
  configTime(2 * 3600, 1 * 3600, "pool.ntp.org", "time.nist.gov");
  delay(2000);
  time_t now = time(nullptr);
  Serial.println(ctime(&now));
}

void loop() {

}
```

### Debug Messages
Expected GMT + 3 to be printed. It actually prints GMT + 2

Cheers

Juppit, 2016-09-12:
For me it's working okay.

Try this:
```
#include <ESP8266WiFi.h>
#include "pass.h" /* ssid=; pass=; */
#include "time.h"

time_t this_second = 0;
time_t last_second = 0;
int dst=3600;

void setup() {
     Serial.begin(19200);
     WiFi.mode(WIFI_STA);
     WiFi.begin(ssid, pass);
     while (WiFi.status() != WL_CONNECTED)
     {
         Serial.print(".");
         delay(500);
     }
     /* let us set: timezone in sec, daylightOffset in sec, 
server_name1, server_name2, server_name3 */
     configTime(7200, 3600, "time.nist.gov", "time.windows.com", 
"de.pool.ntp.org");
     // wait until network has actual time
     while(!this_second)
     {
         time(&this_second);
         Serial.print("-");
         delay(100);
     }
     Serial.println();
}

void loop() {
     time(&this_second); // until time is set, it remains 0
     if (this_second != last_second)
     {
         last_second = this_second;
         if ((this_second % 60) == 0)
         {
             Serial.print(ctime(&this_second));
         }
     }
     delay(500);
}
```

The network needs some seconds to initialize the internal time.
I noticed, that the esp core needs about one hour, to update the time.

Obviously there is a fixed date when DST changes,
I suspect it corresponds to Shanghai time...

On 12.09.2016 at 09:40 wrote tzapu:

Juppit, 2016-09-12:
ATM I have only an old ESP-01 with lots of jumpers...

But I can confirm, parameter dst dosn't have any effect in the last SDK.

This behavior has changed in the last time. I dont know when, but there 
was little change in the SDK.

On 12.09.2016 at 18:06 wrote Peter:
> For me it's working okay.
>
> Try this:
> ```
> #include <ESP8266WiFi.h>
> #include "pass.h" /* ssid=; pass=; */
> #include "time.h"
>
> time_t this_second = 0;
> time_t last_second = 0;
> int dst=3600;
>
> void setup() [...]
e

Booli, 2017-06-21:
I am also experiencing this same behaviour. DST is not being taken into account.

FriedCircuits, 2017-07-26:
Same problem here. I just adjusted the timezone by 1 hr to account for it. 

Seems like at least in time.c this variable gets set and never used again.

s_daylightOffset_sec = daylightOffset_sec;

Granted I haven't looked deeper into the code.

devyte, 2017-10-13:
I still see the code referenced in the previous comment in latest git: the static variable s_daylightOffset_sec 
 is init, set, but never used.
Also, in file time.c function _gettimeofday_r() this looks wrong to me, probably a copy-pasto:
        tp->tv_usec = micros() * 1000;

Juppit, 2017-10-13:
As long as there is no function in the SDK to set DST, sntp_get_current_timestamp will only be correct in Shanghai. :-)
I made a PR to  correct _gettimeofday_r.

mlinaje, 2017-11-04:
A quick search for "s_daylightOffset_sec" in the repository https://github.com/esp8266/Arduino/search?utf8=%E2%9C%93&q=s_daylightOffset_sec&type= shows DST it is still not used.

An interesting function in ../core/sntp.c is:
`void ICACHE_FLASH_ATTR sntp_set_daylight(int daylight)
{
	if (sntp_get_timetype()){
		RTC_DST_SET(daylight);
	}
}`
that seems to write the ESP8266 RTC memory, being RTC_DST_SET defined as:
`do {uint32 value = READ_PERI_REG(RTC_STORE3);`

However, I haven't found where sntp_set_daylight is invoked.

FernandoGarcia, 2018-09-14:
DST offset doesn't work for me too.

tablatronix, 2018-10-18:
#4637

chrisspen, 2019-03-14:
Yeah, this is also happening for me. DST is completely ignored.

Verwalter2017, 2019-04-15:
I got the same issue. Time off by one hour since DST.

d-a-v, 2019-10-01:
SNTP support and example have receive updates, and this thread is old.
Closing but feel free to open a new issue based on git-current sntp code and example.

Current code allows to update sntp startup and update time, 
also once **your timezone is set (with "continent/city"), DST is automatically handled** and automatically updated on transition time without reflash/reboot.
check example.

https://github.com/esp8266/Arduino/pull/6373#issuecomment-536235781
https://github.com/esp8266/Arduino/blob/1aeea124b358c3dc63e1c6fb96376695cf24840b/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino#L42

closed 2019-10-01 by d-a-v