esp8266/Arduino #2508: SPIFFS + ESP8266 + ArduinoIDE + Circular Buffer

MrD2, 2016-09-12:
Hello,

I'm using the SPIFFS File System and i would like to make a buffer to store data when my internet connection is down. I intended to make a circular buffer using for example 32KB of Flash memory...

I made a class using the SPIFFS and taking advantage of the File Like usability of the SPIFFS when reading File type i would use the File.seek() method to point where i write to. The problem is that i can't  after this change where i start to read from (because seek points both the read and write pointers in the File Class)

Can anyone help me? My file configuration is something like this

**********************             At the beginning of the file is the header where the information of
*     Header stuff      *            BUFFER_TAIL, BUFFER_HEAD, BUFFER_SIZE and FREE_POSITIONS are stored.
**********************              
*                              *              I would like to know how i can use the File class READ and WRITE pointers
*      Actual buffer   *              to my advantaging using them to refresh BUFFER_TAIL and BUFFER_HEAD
*           is here        *              eveytime i reset the node for example... power goes down and what not
*                              *
*********************

Thank you for your time

stickbreaker, 2016-09-18:
Just use the file primitives wrapped with your circular buffer code.  You have to implement the _Store()_ and _Recall()_ function.  If the ESP is not subject to power fails just store the Head and Tail values in RAM, use the File.seek() to position the file pointer when necessary, File.position() can be used to identify were the file pointer is currently positioned.
You are going to have to do some thinking on how the data will be divided into packets.  Either have all packets the same size, or use some markers, (crlf) or BOT,EOT or RS,US.  If the data is binary, you might need a size counter before each block. 

Do some research on circular buffer code.  the SPIFFS library can support a circular buffer quite easily.

Chuck.

forthlightning, 2016-09-26:
I'm looking for that same functionality @MrD2! Got a library you care to share? Any help is greatly appreciated.

MrD2, 2016-09-27:
@forthlightning i have something but i never finished.... because of what i tried to explain earlier. @stickbreaker  I implemented the recall and the store functions that use the index in the file (as the own File system uses the index to know from where to read and write).

So, my circular buffer is something like

`class cbuffer{
  public:
    Ctor(); >>> Do some loading stuff
    Store(): ->Saves, head tail, filesize and pos_free to file_buffer header...
    Recall(): ->Recalls previous parameters....
    {
   ???
    }
    queue();
    dequeue();

  private:
    File file_buffer;
    long head;
    long tail;
    long file_size;
    long pos_free;
}`

Now my question is where (after a power failure or for other reason) we have to reassign the pointers used by the file.buffer.write() and file_buffer.read() functions. there are 2 scenarios:
- if **tail is after header** we can f.seek() to the header (in and out stream point to tail) and then read dump until we reach tail (so in == header and out == tail)
- if **header is after tail** you can't do the same because the file wont rewind (or is it???). Can you f.seek to header and then dump read until EOF and the in rewinds to top of file and then u dump seek till tail?? Is this what you are saying @stickbreaker.

In the matters of how the data is divided i only care to store RAW data, so what type of data is stored, and how it is stored is the problem of upper management... you can easily store it using println and then read_line or if the buffer will jump to beginning read until /r/n

When i have something close to a library i can share with you @forthlightning

stickbreaker, 2016-09-27:
@MrD2,
  I think your perception of the two cases is in error. With a ring buffer, the 'head' is always before the tail.  

Let us define the Empty and Full conditions as:
empty: __tail__ == __head__;
full:  next(__head__) == __tail__;

```c++
#define BUFLEN 7
char buffer[BUFLEN];
uint16_t head=0,tail=0;

uint16_t next( uint16_t current){ // calculate next position for head or tail indexes
uint16_t result = (current +1) % BUFLEN; // increment, and wrap around buffer
return result;
}

bool recallByte(byte & b){
bool result = false; // mark as buffer empty
b= buffer[tail]; // read next character, possibly
if(tail!=head){ // character is actually valid
  tail = next(tail); // increment for next read
  result = true; // good character
  }
return result;
}

bool storeByte( byte b){
uint16_t index = next(head);
bool result = false; // mark as failed, buffer full
if(index != tail){ // room exist to store this byte
  buffer[index]=b;
  head = index;
  result = true;
  }
return result;
}
```

With a circular buffer modeled this way, there is no before or after, just equal or not equal.  I think you will find this model will solve you quandary.

Chuck.

MrD2, 2016-09-27:
@stickbreaker 

Head is always ahead of tail (logically).... but in memory space that does not always happen...

stickbreaker, 2016-09-27:
I agree, I just believe the logical view is the only accurate view.  If I try to use before and after, I always get confused.  

My code allows the full buffer to be used. __buffer[head]__ is the last character stored and __buffer[tail]__ was the last character recalled.

How have you decided to handle the full buffer condition?  Do you stop adding or do you eject 'old' data?  

I think you will have to have some block markers, something so that you can eject an entire entry, not just a couple of byte.  I would think partial messages would confuse the upper management.

Chuck.

MrD2, 2016-09-27:
well, if the buffer is only used for one type of data, let's say:
struct some_struct{
   char data[?];
   long type1;
   long type2;
}

you can always try queue() and on return false you can decide what to do.... wait for space, keep going, or you can dequeu() oldest data and enqueue new one....

Other case where data is arbitrarly in size, for example strings, you can use LF CR or for binary data use an header per message 

struct header{
    long length;
    char buffer[];
}

stickbreaker, 2016-09-27:
I agree, you will have to eject an entire message to keep the ring usable. 

My simple example has a hard time with multi-byte blocks or random length strings.  Your examples of fixed length records or self contained packets are more real World.  

You might consider a callback in your object class, who's purpose is to detect the item boundaries.
This user supplied function would be used by the __recall()__ function to identify where the next element ends.  The helper functions would walk through the ring building the return value until they have matched the count returned by the callback.

This sound like it could be useful addition to the ESP project.

Good luck, the devil is in the details ;)

Chuck.

thehellmaker, 2016-09-30:
This is good. I had a similar discussion with @igrr and @me-no-dev on gitter. @MrD2 Do you have anything concrete? If nto i'll go ahead and write my own and get back to you guys once i am done.

MrD2, 2016-09-30:
@thehellmaker I have something started but i have some problems yet...
1. I can't read binary data from header
2. i can't read binary data from buffer
3. Storing header is a bit tricky because it lags a lot the rest of the program (if i for example store it each time i change something in the buffer... one way to fix this is to save only from time to time [with some timer] or to save if i changed more than x bytes in the buffer [a ram counter]). 

I can send to you what i have and you can check it.... but there are still some functions that i did not test.
PM me with your e-mail?

D. Guerra

MrD2, 2016-10-03:
Well, i have something but it's not working properlly...
If it were C would be much easier but this Arduino language can't read chuncks of data like i want... And it seems that opening the file with a+, w+ or r+ as different results and inpacts the knowledge of where the buff is and where it writes/reads from... which is stange because i walways issue seek before any kind of IO.

Here are some functions....

`
#ifndef LOGGER_H
#define LOGGER_H

#include <stdint.h>
#include "FS.h"

#define FS_PAGE_SIZE  128
#define NUM_PAGES 256
#define RING_BUFFER_SIZE (NUM_PAGES * FS_PAGE_SIZE)

// Uncomment to enable printing out nice debug messages.
#define LOGGER_DEBUG
// Define where debug output will be printed.
#define DEBUG_PRINTER Serial1

#if (RING_BUFFER_SIZE & (RING_BUFFER_SIZE - 1)) != 0
#error "RING_BUFFER_SIZE must be a power of two"
#endif

#define RING_BUFFER_MASK (RING_BUFFER_SIZE-1)

// Setup debug printing macros.
#ifdef LOGGER_DEBUG
  #define DEBUG_PRINT(...) { DEBUG_PRINTER.print(__VA_ARGS__); }
  #define DEBUG_PRINTLN(...) { DEBUG_PRINTER.println(__VA_ARGS__); }
#else
  #define DEBUG_PRINT(...) {}
  #define DEBUG_PRINTLN(...) {}
#endif

class Logger {
  public:
    Logger(const char *file_name);
    Logger(const char *file_name, long len);
    ~Logger();
    /**
     * @brief   Adds a byte to the ring buffer.
     * @param   data  The byte to place.
     * @retval  true  if success.
     * @retval  false otherwise.
     */
    bool queue(uint8_t data);
    /**
     * @brief   Adds an array of bytes to the ring buffer.
     * @param   data  A pointer to the array of bytes to place in the queue.
     * @param   len   The size of the array.
     * @retval  true  if success.
     * @retval  false otherwise.
     */
    bool queue(const uint8_t *data, long len);
    /**
     * @brief   Returns the oldest byte in the ring buffer.
     * @param   data  A pointer to the location at which the data should be placed.
     * @retval  true  if data was returned;
     * @retval  false otherwise.
     */
    bool dequeue(uint8_t *data);
    
    /**
     * @brief   Returns the <em>len</em> oldest bytes in the ring buffer.
     * @param   data  A pointer to the array at which the data should be placed.
     * @param   len   The maximum number of bytes to return.
     * @retval  true  if data was returned; 
     * @retval  false Otherwise.
[... 149 lines cut ...]
    }    
  } else{
    DEBUG_PRINTLN("LOGGER: cant read amount of data requested");
    return false;
  }
  return true;
}
//___________________________________________PRIVATE_____________________________

void Logger::store(){
Serial1.print("THIS IS MY POSITION" + String(this->file_buffer.position()));
  this->file_buffer.seek(0, SeekSet);
  this->file_buffer.printf("%ld|%ld|%ld|%ld|\r\n",this->head_index, this->tail_index, this->file_size, this->pos_free);
}

void Logger::recall(){
  char term = '|';
    
  this->file_buffer.seek(0, SeekSet);  
  this->head_index  = this->file_buffer.readStringUntil(term).toInt();
  this->tail_index  = this->file_buffer.readStringUntil(term).toInt();
  this->file_size   = this->file_buffer.readStringUntil(term).toInt();
  this->pos_free    = this->file_buffer.readStringUntil(term).toInt();
}
`

forthlightning, 2016-10-03:
I had some similar issues around keeping track of the read/write heads of the File object. I worked around it by keeping track separately of the index that each was supposed to be, and then using f.seek() to move to the appropriate location.

One more thing I ran into was that I was using f.println() to put my message into the FS, so there was an extra '\n' that had to be read out of the buffer in order to get the following messages.

devyte, 2017-10-13:
Not the right place to have this discussion. Closin per #3655 .

closed 2017-10-13 by devyte