esp8266/Arduino #2509: esp8266 station connected on AP causes wifi performance drop

asmeureanu, 2016-09-12:
### Basic Infos

#### Hardware
Hardware:	    NodeMCU 1.0 ESP-12E
Core Version:      	2.2.0 and  2.3.0-rc2

### Description
I connect an ESP12E flashed as an station with IP assigned over DHCP with any sample wifi program (HelloServer / basic http server) the performance of the network drops.  The presence of the device on  the wifi network disrupts. E.g basic http transfer via Safari on my iphone via the same AP towards internet becomes very sluggish. As soon as I disconnect the device from the network the transfer returns to normal.

I have tried it with multiple boards to ensure that the hardware is not the issue.

I do not experience this issue with the rest of the ESP12E NodeMCU devices i own running lua. 

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP12E )
Flash Size: 		4MB
CPU Frequency:		80Mhz

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "AP-55";
const char* password = "AP55wireless";

ESP8266WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
}

WereCatf, 2016-09-12:
> flashed as an station
Mmmno. You just call WiFi.begin() without specifying mode beforehand, which means it's acting as both an AP and a STA, and that is likely the cause of your issue. Call `WiFi.mode(WIFI_STA);` before WiFi.begin() to place it in a STA-mode only.

asmeureanu, 2016-09-14:
Hi,

Thank you very much! That was the issue calling WiFi.mode(WIFI_STA);  Solved the issue.

It is bad that a such a basic example HelloServer picked from the Arduino IDE does not point this. 
not even with comment on this.  I suggest fixing this in 

https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/HelloServer

I will open a separate ticket for this.

closed 2016-09-14 by asmeureanu