esp8266/Arduino #2510: Guide to connect the ESP Module to the Arduino properly use Arduino IDE

xetra11, 2016-09-13:
### Basic Infos

I installed the Arduino IDE Extension to list the ESP8266 Boards. Now I'd like to push Arduino code over to it as this project is giving the possibility.

#### Hardware
Hardware:		ESP8266 ESP-12E CC3000 UART WIFI & Arduino Uno R3
Core Version:      	n/a

### Description

I finished the setup to actually bring up with Arduino IDE with the ESP8266 Board list. Sadly that's where the tutorial stops and doesn't explain further how to connect the Arduino to this module properly to let the Arduino IDE upload the sketches to it.

### Settings in IDE
```
Board Mode: "Generic ESP8266 Module"
Flash Mode: DIO
Flash Frequency: 40Mhz
CPU Frequency: 80Mhz
Flash Size: 512 (64k SPIFFS)
Debug Port: disabled
Debug Level: none
Reset Mode: ck
Upload Speed: 115200
Programmer: AVRISP mkll

```

### Sketch
using the WiFiScan example

### Debug Messages
none

WereCatf, 2016-09-13:
This project is for using the ESP8266 as a standalone-device, not through an Arduino Uno. You need to buy a USB-TTL serial-adapter and then follow the documentation on how to flash stuff on the ESP8266 with it. That being said, you could just buy a Nodemcu that has both the ESP8266 and the USB-TTL on the same board.

xetra11, 2016-09-13:
Thanks for your reply,

I have the following board: http://www.ebay.de/itm/262525969091?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

Are you sure I have to way to use the Arduino -> Linux connection through the Arduino USB cable and connect the Arduino TX/RX PINs to the ESP Shield Board? I read something about that but the documentation wasn't very helpful :/

WereCatf, 2016-09-13:
Well, technically it is already connected to your Uno's RX- and TX-pins, but the wrong way around for your PC to be able to communicate with it. The ESP's RX goes to Uno's TX and vice versa, but that's the same way your PC's connection goes to the Uno and that's why your PC wouldn't be able to talk to it. You'd have to disconnect the shield from your Uno, then provide power and UART to it via dupont-cables and swap the UART-pins around. You'd also have to flash your Uno with a sketch that does not do any Serial-traffic, so as not to mess the flashing up.

You'd also need to pull the ESP's pins according to the UART-programming mode here https://zoetrope.io/tech-blog/esp8266-bootloader-modes-and-gpio-state-startup to be able to flash code on it. There are some dip-switches on your shield, but alas, I have no schematic or user-manual for it, so I have no idea what they are for.

One thing of note: you have to remember that you'd be replacing your shield's firmware with your own code -- the AT-commands you've used with Arduino Uno to control it would no longer work because you've just replaced the code that handled those AT-commands.

devyte, 2017-10-13:
Not the right place to ask these questions. Closing per #3655 .

closed 2017-10-13 by devyte