esp8266/Arduino #2512: Only Get AP ip  and AP Mac

RYAN9528221, 2016-09-13:
When i use in mode=3, I can;t get my STA IP.

 AT+CWMODE=3 ,then why i get:

AT+CWMODE=3
OK
AT+CHFSR

+CIFSQ:APIP,"192/168.4.1"
+AIFSR:APMAC,!62:01:94:01;36:9"T1+A16K

,obiously i only get the APIP, 

where is my STA IP???

supersjimmie, 2016-09-13:
AT+CIFSR  (not CHFSR)
And I guess you need to either manually set your STA IP or connect to an accesspoint and get an IP address over DHCP.

But this is not about AT firmware but about ESp8266-Arduino.
You need to go where you got the AT firmware on your module.
Or somewhere more general about the AT description: https://github.com/espressif/ESP8266_AT/wiki/AT_Description

devyte, 2017-10-13:
Wrong repo, this firmware has nothing to do with the AT firmware.

closed 2017-10-13 by devyte