esp8266/Arduino #2514: Software-Serial with GPS and Servo Library

sleepingshorty, 2016-09-13:
Please fill the info fields, it helps to get you faster support ;)

### Basic Infos
Arduino 1.6.11
Windows 10

#### Hardware
Hardware:			WeMos D1
Core Version:      	2.3.0

### Description
Problem description

GPS is attached to PIN D5 and D6. When I attach a Servo to a Pin (I used D4 and D3) it starts a slow jitter like 3 times a second.

### Settings in IDE

Module:  			WeMos D1 R2 & Mini
Flash Size: 		4/3
CPU Frequency:		80Mhz
Flash Mode:			NA
Flash Frequency:	NA
Upload Using:		OTA and Serial
Reset Method:		?

### Sketch

```cpp
     static const int RXPin = D5, TXPin = D6;
     static const uint32_t GPSBaud = 38400;
     TinyGPSPlus gps;
     SoftwareSerial ssGps(RXPin, TXPin);
#include <Arduino.h>

void setup() {
  ssGps.begin(GPSBaud);
  sailServo.attach(D4);
}

void loop() {
     while (ssGps.available())
         gps.encode(ssGps.read());}
```

### Debug Messages

```
That's it and the Servo starts to jitter.
```

iliis, 2017-06-04:
Can confirm, I'm having the same issue.
Software Serial library from https://github.com/plerup/espsoftwareserial, version 3.2.3

devyte, 2017-10-12:
@sleepingshorty I'm going to assume that the gps sends updates at about 3 times per second?
At 38400 baud, receiving one byte takes about 260us, assuming my quick sloppy math is correct. Considering that both the servo and softwareserial libs are emulated in software, the best case you can expect is a jitter of 260us in servo, which can happen while softwareserial is receiving a byte.
The jitter here is 700us. The remaining 440us could be overhead somewhere else.
I'm not sure what, if anything, can be done here.

devyte, 2017-10-12:
@igrr do you have any insight here?

igrr, 2017-10-12:
I think we have modified PWM implementation to use NMI interrupt, so I'm surprised that this still is an issue. I would have expected the reverse, that is SoftwareSerial being broken by PWM.
No immediate clues, need to debug this.

earlephilhower, 2019-01-16:
This should be fixed by a combination of #5578 and #4640.

Before Servo was using timer1 directly, not PWM, so the PWM NMI bit that @igrr mentioned wouldn't even be used.  Now, Servo uses the waveform generator which is also NMI, and should be invariant of IRQ disable by SW Serial.

Closing for now, but if someone has the same issue with the GIT head, please do open up a new issue.

closed 2019-01-16 by earlephilhower