esp8266/Arduino #2515: atof() does not grok sci notation

pidloop, 2016-09-14:
### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			Adafruit HUZZAH ESP
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {
    Serial.begin(9600);
    while (!Serial);

    char *sci = "1.0e-7";
    Serial.println (sci);
    Serial.println (1e7*atof(sci));

}

void loop() {

}
```

### Debug Messages

```
messages here
```

Output is:

1.0e-7
10000000.00

should be
1.0e7
1.00

pidloop, 2016-09-14:
I also tried

   Serial.println (strtod(sci,(char**)NULL));

but same result.

I forgot to mention, both tests also had #include <stdlib.h>

devyte, 2017-10-13:
@pidloop is this issue still valid with latest git? still valid with #3215 ?

pidloop, 2017-10-13:
Arduino Board Manager says I am running ESP8266 Community version 2.3.0 and yes, this error report is still valid.

earlephilhower, 2019-01-16:
This works in git head, closing.

closed 2019-01-16 by earlephilhower