esp8266/Arduino #2516: Example ESP8266WebServer/examples/HelloServer does not call   WiFi.mode(WIFI_*);

asmeureanu, 2016-09-14:
Depending on the use case; by not calling  WiFi.mode(WIFI_STA); you create as well an additional AP with the same SSID and password as you original AP that you want to connect to.

This causes issues for the other devices on you wifi network, ranging from disconnect to very bad wifi performance. 

Please adapt the example to point that the user should chose one the mode he needs to use.

devyte, 2017-10-12:
@asmeureanu no basic info, no IDE settings.
I don't understand the issue here. I can run the example just fine, I have never seen "an additional AP with same SSID and password".
Is the issue still relevant with latest git? if so, please provide a better explanation.

devyte, 2018-11-08:
The mentioned example already calls WiFi.mode(WIFI_STA). 
Closing.

closed 2018-11-08 by devyte