esp8266/Arduino #2518: can't connect to my wifi router anymore!

mushfiq-cse05-buet, 2016-09-14:
hello everyone, i'm facing a brand new problem that didn't happen to me before, a week ago i have changed my wifi router's settings and after that my esp8266(esp-01) can't connect to my wifi router anymore ??? can anybody tell me whats the problem???

WereCatf, 2016-09-15:
> can anybody tell me whats the problem???

"I changed stuff, but I'm not telling you what I changed and I just want you to magically know what's wrong." -- Mmmno. How about telling what settings you changed, or reverting your changes?

devyte, 2017-10-12:
If this issue is still valid, I suggest doing a fill reset of your router, and configuring from scratch.
Not the right place to ask for such help, closing per #3655 .

closed 2017-10-12 by devyte