esp8266/Arduino #2519: No action with HTML  [Toggle Pin11] or [Toggle Pin12] or [Toggle Pin13]

RYAN9528221, 2016-09-15:
I want to use the HTML to control My Arduino UNO LEDS(pin11,pin12,pin13),
 found that I can use chorme to key in 192.168.106.1:8888 ,then it show me:
-------------------------------------------------------------------------------------------
Enter case 6
ENTER WEB1
ENTER WEB2
Get webpage signal, analyzing...
Enter case 7
Turn Pin-539:1!
SEND: AT+CIPCLOSE=0

Enter case 6
Enter case 6

-----------------------------------------------------------------------------------------------
but, when use html code:

----------------------------------------------------------------------------------------------
<html>
	<head>
		<title>ESP8266 LED Control</title>
	</head>
	<body>
	
	
	
	<button id="11" class="led">Toggle Pin 11</button> 
	<button id="12" class="led">Toggle Pin 12</button> 
	<button id="13" class="led">Toggle Pin 13</button> 
		
	<script src="jquery.min.js"></script>
	<script type="text/javascript">
		$(document).ready(function(){
			$(".led").click(function(){
				var p = $(this).attr('id'); // get id value (i.e. pin13, pin12, or pin11)
				// send HTTP GET request to the IP address with the parameter "pin" and value "p", then execute the function
				$.get("http://192.168.1.106:1:8888/", {pin:p}); // execute get request
			});
		});
	</script>
	</body>
</html>
-----------------------------------------------------------------------------------------------------.
I click button  [Toggle Pin11] or [Toggle Pin12] or [Toggle Pin13],

it have no action..........................

is my Html code wrong???????

and next is my Arduino sourece code:
--------------------------------------------------------------------------------------------------
#include <SoftwareSerial.h>
#include <Timer.h>
SoftwareSerial DRX(4,5);

#define SSIDWORD "hinet4-14"
#define PASSWORD "033251196"
Timer t;

void setup() {
  Serial.begin(115200);
  DRX.begin(115200);
  pinMode(10,OUTPUT);
  digitalWrite(10,LOW);
  pinMode(11,OUTPUT);
  digitalWrite(11,LOW);
  pinMode(12,OUTPUT);
  digitalWrite(12,LOW);
  t.every(500, Wifi_main);
}

void loop() {
  t.update();

}
int Wifi_State,delaytime;
String cmd;
void Wifi_main()
{
  switch (Wifi_State) 
  {
    case 0:
    sendDebug("AT");
    if(Loding("Sent AT"))
    {
      Wifi_State++;  
  //    digitalWrite(11,HIGH);
      Serial.print("\n Next to case1\n");
    }
    break;
    case 1:
   //   sendDebug("AT+CWMODE=3");
    //  if(Loding("AT+CWMODE=3"))
      {
         Wifi_State++;  
          Serial.print("\n Next to case2\n");
      }
   case 2://--------------------------------------------------Wifi connect
   Serial.print("Enter case 2\n");
//    digitalWrite(12,HIGH);
      cmd="AT+CWJAP=\"";
      cmd+=SSIDWORD;
      cmd+="\",\"";
      cmd+=PASSWORD;
      cmd+="\"";
  Serial.print("cmd comment= ");
  Serial.print(cmd);
      sendDebug(cmd);
      Wifi_State++;
      if (Loding("Wifi connect")){ 
        Wifi_State++; //next setting wifi mode
 //       digitalWrite(12,HIGH);
      }
      else{
        Wifi_State--;
 //       digitalWrite(13,HIGH);
        }
    break;
   case 3://--------------------------------------------------Get ip address 數據機"http://192.168.0.6:8888/"  手機熱點" http://172.20.10.8:8888/pin="  <=== ip 
    Serial.print("Enter case 3\n");
      sendDebug("AT+CIFSR");
      if (Loding("Get ip address")){ //delay 1 sec
        Wifi_State++; //next setting wifi mode
      }
      else{Wifi_State--;}
    break;

      case 4://--------------------------------------------------configure for multiple connections
      Serial.print("Enter case 4\n");
      sendDebug("AT+CIPMUX=1");
    //  if (Loding("Set CIPMUX")){ 
          Wifi_State++; //next setting wifi mode
    //  }
  //    else{Wifi_State--;}
    break;

  case 5://--------------------------------------------------turn on server on port 8888
     Serial.print("Enter case 5\n");
      sendDebug("AT+CIPSERVER=1,8888");
      Wifi_State++; //next setting wifi mode
    break;

    case 6:// check if the esp is sending a message 
    Serial.print("Enter case 6\n");
   //  digitalWrite(11,HIGH);
      if(DRX.available()) {
         Serial.print("ENTER WEB1\n");
   //      digitalWrite(11,HIGH);
          Serial.print("ENTER WEB2\n");
        if(DRX.find("+IPD,")){
          Serial.println("Get webpage signal, analyzing...");
          Wifi_State++;
          }
      }
    break;

  case 7:// wait for the serial buffer to fill up (read all the serial data)
  Serial.print("Enter case 7\n");
        delay(10);
        int connectionId =  DRX.read()-48; // subtract 48 because the read() function returns 
        // the ASCII decimal value and 0 (the first decimal number) starts at 4
        DRX.find("pin="); // advance cursor to "pin="
        int pinNumber = (DRX.read()-48)*10; // get first number i.e. if the pin 13 then the 1st number is 1, then multiply to get 10
        pinNumber += (DRX.read()-48); // get second number, i.e. if the pin number is 13 then the 2nd number is 3, then add to the first number
      //539 1
        digitalWrite(pinNumber, !digitalRead(pinNumber)); // toggle pin    
     //    digitalWrite(12, HIGH);
         
        // make close command
        String closeCommand = "AT+CIPCLOSE="; 
        closeCommand+=connectionId; // append connection id
        closeCommand+="\r\n";
        Serial.print("Turn Pin");
        Serial.print(pinNumber);
        Serial.print(":");
        Serial.print(digitalRead(pinNumber));
        Serial.println("!");
        sendDebug(closeCommand); // close connection
        Wifi_State--; //next setting wifi mode
     break;

    
  }

}
String get_response() {  //get esp responce without "Serial.find()".
  String response="";  
  char c; 
  while (DRX.available()) {  
    c=DRX.read();
    response.concat(c);   
    delay(10);  
    }
  response.trim();  
  return response;
}
boolean Loding(String state){
  String response=get_response();
  for (int timeout=0 ; timeout<30 ; timeout++)
  {
    if(response.indexOf("OK") != -1 || response.indexOf("no change") != -1) 
    {
        Serial.print(" OK ! Loading Complete");
        if(timeout>=2){
          Serial.print("This commend waste ");
          Serial.print(timeout/2);
          Serial.println(" sec.");
        }
        else Serial.println("");
        return 1; 
        break;
    }
    else if(timeout==29){ //after 15 sec for wait esp without responce.
      Serial.print(state);
      Serial.println(" fail...\nExit2");
      return 0; 
    }
    else{
      response=get_response(); //reget
      if(timeout==0)
        Serial.print("Wifi Loading.");
      else
        Serial.print(".");
      delay(500);
    }
  }
}
void sendDebug(String sent_cmd)
{
    Serial.print("SEND: ");
    Serial.println(sent_cmd);
    DRX.println(sent_cmd);
}

RYAN9528221, 2016-09-15:
I resolute it by my self, this question is because i didn't download JQuery Lib.
After I downlaod,
jquery.min.js

and it is work!
But the next issue is the signal of antenna is too bad.
Maybe my 3.3 V is not  stabel.

devyte, 2017-10-12:
Not the right place to ask for help for your particular project. Closing per #3655 .

closed 2017-10-12 by devyte