esp8266/Arduino #2521: SPIFFS wdr and erases flash or isn't mountable

bhorn, 2016-09-15:
### Basic Infos

#### Hardware
Hardware: ESP-01
Core Version: 2.1.0-rc2

### Description

I'm trying to mount the SPIFFS but run into some troubles. I managed to extract two different problems:

#### 1
When I set my IDE to any Flash-Size and run a snippet with only `SPIFFS.open()` and `.close()` it seems like the WDR interrupts the script. When I reboot the ESP it looks like the flash got erased:

```
0, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
BÎZ-V�D	“%ÎÚw†ò‹ÐY�xKÁÃTÍÚÅ <---wdr suddenly talks in 118200 instead of 74880
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

ets_main.c 
```
If I remove the SPIFFS.open(), everything works fine in any Flash-Size.

#### 2
When I go to boards.txt and "hack" some custom settings in it for 8M because I have a 8M flash, the sketch runs fine but `.open()` failes.
When I try to `.open()` a second time the debug-output shows that it can't mount the FS 
```
SPIFFSImpl: mounting fs @100000, size=700000, block=2000, page=100
_spif_read(80) addr=1000fc size=2 ab=1000fc ae=1000fc
SPIFFSImpl: mount rc=-1
_spif_erase addr=100000 size=4096 i=0
_spif_erase addr=101000 size=4096 i=0
_spif_write(118) addr=1000fe size=2 ab=100100 ae=100100
SPIFFS_format: rc=-1, err=-10027
```

boards.txt 
```
generic.menu.FlashSize.8M=8M (7M SPIFFS)
generic.menu.FlashSize.8M.build.flash_size=1M #Nothing else works...
generic.menu.FlashSize.8M.build.flash_ld=eagle.flash.8m.ld
generic.menu.FlashSize.8M.build.spiffs_start=0x100000
generic.menu.FlashSize.8M.build.spiffs_end=0x800000
generic.menu.FlashSize.8M.build.spiffs_blocksize=8192
generic.menu.FlashSize.8M.upload.maximum_size=1044464
```

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/8MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp
#include "FS.h"
#include <Arduino.h>

void setup() {
  Serial.begin(74880);
}

void loop() {
  if(SPIFFS.open())
    Serial.println("yay!");
  else
    Serial.println("fak.");
  SPIFFS.end();
}
```

WereCatf, 2016-09-15:
Um, I hate to rain on your parade, but... I'm not seeing a call to `SPIFFS.begin();` in your sketch.

bhorn, 2016-09-15:
Sorry! Copy&Paste fault. Actualy there is a begin-call as first call in loop().
I'll update the Sketch.

WereCatf, 2016-09-15:
You have an ESP-01 with 8MiB flash? Have you replaced the original one with another one? ESP-01's are supposed to have 512KiB or 1MiB. If the flash-chip has the text "25Q40" on it then it's 512KiB and if it says "25Q80" then it's 1MiB.

bhorn, 2016-09-15:
Ok thats odd, never changend the Flash. The bootloader told me multiple times 8MB and even the FlashCheck Sketch told me everything is fine. I've checked the prints on the memory, it says 25Q but then some other (sadly not that readable anymore) letters.
I'll double check tomorrow!

WereCatf, 2016-09-15:
I am 99% sure you got your flash-size wrong, so yeah, you should try hard to read what it says.

bhorn, 2016-09-21:
So I didn't manage to get the correct flash-size. It says something like 25QAA on my Flash. So I desoldered it and put a 25Q80 on it. Works like a charm with 512+512. 
Consider this Issue as Hardware-Fault.

closed 2016-09-21 by bhorn