esp8266/Arduino #2522: Assert.h not 'firing' on somewhat simple expression

airfling, 2016-09-15:
### Basic Infos
Hi! I know that error must be sitting in front of the PC, but I'm quite helpless. I've been working with assert() quite a long time and it works well. But when writing for a small tutorial, I've been experiencing a very strange behaviour.

### Description

Using assert(0) _fails_ just great and the ESP gives me a stacktrace etc. But the following sample just will not work. Even worse: After the assert-statements in 'turnServo()' the if-statement fails, too. It looks like, that assert somewhat tries to fail, but fails on that and scrambles the stack(?)/processor(?).

Any idea would be great. Thank you in advance.
Holger

### Sketch
Main.cpp:

```
#include <Arduino.h>
#include <assert.h>

void turnServo(int angle); // foward declaration required by c++

int currentAngle = 0;

void setup() {
	Serial.begin(9600);
	delay(2000);
}

void loop() {
	currentAngle += 10;
	turnServo(currentAngle);
	delay(200);
}

void turnServo(int angle) {

	if(angle>360) {
		Serial.println("Assert MUST work! #1");
	}
	assert(angle>0);
	assert(angle<360);

	if(angle>360) {
		Serial.println("Why don't we get here? It's the same boolean expression as above?");
	}

	Serial.print("Turning to ");
	Serial.println(angle);
}
```

#Debug output:
```
...
Turning to 310
Turning to 320
Turning to 330
Turning to 340
Turning to 350
Turning to 360
Assert MUST work! #1
Turning to 370
Assert MUST work! #1
Turning to 380
...
```
It looks like, that the expression in the assert is simply ignored.

#### Hardware
Hardware:			ESP-12
Core Version:      	?2.1.0-rc2? (Adafruit Huzzah? Sorry)

### Settings in IDE
IDE: Platform.IO
Module:  			Adafruit HUZZAH ESP8266
Flash Size: 		4MB
CPU Frequency:		80Mhz

airfling, 2016-09-15:
PERFECT!!! That was the reason (and for me the command to buy JTAG :-)

Now, everthing works fine!

Thank you very much

closed 2016-09-15 by airfling

airfling, 2016-09-15:
@igrr Sorry, I'm a terrible newbie to 'not-working-alone'. Is there a help-page / how-to on committing changes / improvements to this repository? I mean: You solved the bug, so why should you improve the repository? Would it be possible for me to propose this change (and maybe later on to help on other issues on the platform?)

krzychb, 2016-09-15:
@airfling,

1. Check ["Contributing"](https://github.com/esp8266/Arduino#contributing)
2. Press "Fork"
[image]
3. Make the changes and submit a [pull request](https://help.github.com/articles/about-pull-requests/)

This is newbie friendly :smile: :+1: