esp8266/Arduino #2525: Error certificate validation

Regirv, 2016-09-16:
### Basic Infos

#### Hardware
Hardware:			ESP-07
Core Version:      	2.3.0-rc2

### Description

Unable to verify the certificate. Error:
pos:0 len:62 invalid hex sequence: 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "wifi";
const char* password = "pass";
char server[] = "host";
const int port = 443;
const char* fingerprint = "‎‎‎10 27 73 B8 C5 5A 2B 41 4A 38 1A EA 71 21 ED D7 6C 19 09 13";

void setup() {
Serial.begin(115200); 
  Serial.setDebugOutput(true); 
   delay(10);
  WiFi.begin(ssid, password); 

while (WiFi.status() != WL_CONNECTED) {
 Serial.println("OK");
  }
erial.print("connecting to ");
  Serial.println(server);

  WiFiClientSecure client;
  if (!client.connect(server, port)) {
    Serial.println("connection failed");
    return;
  }

  if (client.verify(fingerprint, server)) {
    Serial.println("certificate matches");

  }
  else {
    Serial.println("certificate doesn't match");

  }
}

void loop() {

}
```

### Debug Messages

```
connected with wifi, channel 6
dhcp client start...
ip:192.168.1.33,mask:255.255.255.0,gw:192.168.1.1

WiFi connected
IP address: 
192.168.1.33
connecting to host
:ref 1
ssl/tls1.c:545 malloc 6864, left 30544
please start sntp first !
State:	sending Client Hello (1)
:wr
:sent 77
:ww
:rn 1034
:rd 5, 1034, 0
:rdi 1034, 5
:rd 74, 1034, 5
:rdi 1029, 74
State:	receiving Server Hello (2)
:rd 5, 1034, 79
:rdi 955, 5
:rd 941, 1034, 84
:rdi 950, 941
State:	receiving Certificate (11)
crypto/bigint.c:1072 realloc 1032, left 26152
:rd 5, 1034, 1025
:rdi 9, 5
:rd 4, 1034, 1030
:rdi 4, 4
:c0 4, 1034
State:	receiving Server Hello Done (14)
crypto/bigint.c:1072 realloc 1024, left 22760
State:	sending Client Key Exchange (16)
:wr
:sent 267
:ww
:wr
:sent 6
:ww
State:	sending Finished (16)
:wr
:sent 69
:ww
:rn 75
:rd 5, 75, 0
:rdi 75, 5
:rd 1, 75, 5
:rdi 70, 1
:rd 5, 75, 6
:rdi 69, 5
:rd 64, 75, 11
:rdi 64, 64
:c0 64, 75
State:	receiving Finished (16)
pos:0 len:62 invalid hex sequence: ??
certificate doesn't match
:wr
:sent 53
:rcl
:abort
:ww
Alert: close notify
:ur 1
:del

```

pajasevi, 2016-11-27:
Hey guys, I'm having exactly the same problem. Could someone take a look at this please?

pajasevi, 2016-11-27:
@Regirv Issue resolved... It seems that when I copied SHA1 fingerprint to the sketch editor, it somehow is in wrong charset... So when I retyped fingerprint manually, it works.

Gosh, spent like 4 hours with this...

devyte, 2017-10-12:
@Regirv is this issue still valid with latest git? There were some commits recently.

devyte, 2018-07-12:
Closing due to lack of feedback and age.

closed 2018-07-12 by devyte