esp8266/Arduino #2526: tcp fail on  modem-sleep

ali-libre, 2016-09-16:
without modem-sleep pocket send successfully but
```
WiFi.forceSleepBegin();
delay(1);
WiFi.forceSleepWake();
Serial.println("modem waked");
delay(1);
```
with delay or extra delay or without dely
tcp socket will fail, one by one not more or less
have i to ignore power saving..?
generic-esp...
no-debug-info
in server side faile with 'connection reset by peer'

WereCatf, 2016-09-16:
How do you expect TCP-traffic to work when WiFi is put to sleep? Of course it will fail.

ali-libre, 2016-09-16:
i'm sorry to not explain truely.
after powerup/wakeup tcp will fail.
in  loop one-time successfull and one-time will fail
exactly in tow, one will fail "not more o less" only 1/2 socket connect & 1/2 will fail

devyte, 2017-10-12:
@ali-libre connection is probably dropped at remote end during sleep, or due to the sleep-wakeup. you have to reconnect wifi, and reconnect socket.
Off-topic, closing per #3655 .

closed 2017-10-12 by devyte