esp8266/Arduino #2531: connecting RST to GPIO 16 for  automatic wake-up and using GPIO 2 for external wake-up  with pull up

dasarathsbricks, 2016-09-16:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

middelink, 2016-09-18:
And what is the issue (for the Arduino lib) here exactly?

dasarathsbricks, 2016-09-19:
not related to Arduino lib wrong thread  my Bad :)

StuckInLoop, 2016-09-27:
dasarathsbricks, are you able to get the external wake up working? if so, can you share on how to get it working? thanks

devyte, 2017-10-12:
Non-issue, closing.

closed 2017-10-12 by devyte